Alcolawl/LibreHardwareMonitor #1: Fan headers showing RPM but cannot control

WesFarley, 2025-01-14:
As posted in: https://github.com/LibreHardwareMonitor/LibreHardwareMonitor/issues/1513

Using fork files of LHM. Currently showing RPMs for system fan headers 1-5 which is what I have installed, but during the speed sensor pairing process, no fan responds to speed adjustments.

Win 11 Pro
MSI X870 Tomahawk Wifi Mobo

[image]

Alcolawl, 2025-01-14:
I've found the offending code. In my efforts to optimize the function responsible for writing the fan curve for the target fan, I broke the comparison logic, which needs to be revisited. Specifically, commit [5cf5eab](https://github.com/LibreHardwareMonitor/LibreHardwareMonitor/commit/5cf5eab9ddf250ecac3e1844f061fe2f3bdf8c1d). For now, I've fallen back to an earlier method that I've verified is working and will get it into the next release shortly after I get back from the gym :D

WesFarley, 2025-01-14:
I'm heading to the garage gym as well!
Thank you sir.

WesFarley, 2025-01-14:
I'm heading to the garage gym as well! Thank you sir.

-------- Original Message --------

Alcolawl, 2025-01-14:
I re-created your issue on my end, and tested this new build with my PC (we have the same motherboard) and things are working properly again. Give this a shot and let me know how it works for you. 
[LHM_MSI_Z890NeedsLoveToo_v2.zip](https://github.com/user-attachments/files/18416876/LHM_MSI_Z890NeedsLoveToo_v2.zip)

WesFarley, 2025-01-14:
Functionality restored!  Much appreciated man.  First time using fancontrol so there may be other hardware or operator issues.  I have 3 fans on 2 different headers that arent stopping full at a 0% even after forced apply.  Some fans in the same bank/same header are fully stopping however.  I think the 3 problem children are ones I pulled off an AIO so it wouldnt surprise if they were configured to never fully shut off.

Alcolawl, 2025-01-15:
You may have to manually calibrate those fans. When I went through this testing process, I had to manually set some of my fans to 70, 80, 80, 100% and wait for the fan to settle on it's true RPM for that PWM %. The auto calibration is too impatient. Try manually calibrating the fans in question and set the slider to 0%, the fans should stop spinning.

WesFarley, 2025-01-15:
Happy to close the issue itself :)
Still cant get those particular fans to stop.  I'm not sure its calibration, 2 of 3 fans connected to the same header are at min speed, while 1 completely stops.

Alcolawl, 2025-01-15:
> Happy to close the issue itself :) Still cant get those particular fans to stop. I'm not sure its calibration, 2 of 3 fans connected to the same header are at min speed, while 1 completely stops.

Let me know if you wind up making headway with that, that's weird. Perhaps they're DC fans instead of PWM? idk.

Anyways, glad we got it sorted out. Actually gave me a chance to work more on the optimization code and I fixed that too. 😄

closed 2025-01-15 by Alcolawl