Alcolawl/LibreHardwareMonitor #2: No fan rpm/control compatibility on MSI MPG Z890 Carbon Wifi

Dr-Mugg, 2025-01-15:
**Describe the bug**
SYS fan RPM readout or control not available on MSI MPG Z890 Carbon Wifi in LHM or Fancontrol with modified files.
CPU and PUMP works fine in both applications.

**To Reproduce**
1. Replace files with the supplied ones in Fancontrol folder or start LHM fork version
2. No RPM readout and no adjustability.

**Expected behavior**
SYS-fans should be readable in FC/LHM and adjustable like CPU and PUMP. Hwmonitor can track all connected fans.

**Desktop (please complete the following information):**
 - OS: Win11
- Motherboard LHM "MSI MPG Z890 CARBON WIFI (MS-7E17)"
- Motherboard Hwmonitor "MPG Z890 CARBON WIFI (MS-7E17) (0x00000745 - 0x5276C320)"

**Additional context**
Log from both Hwmonitor and LHM fork:
[LibreHardwareMonitor.Report.txt](https://github.com/user-attachments/files/18429973/LibreHardwareMonitor.Report.txt)
[HWMonitor.txt](https://github.com/user-attachments/files/18429974/HWMonitor.txt)

Alcolawl, 2025-01-17:
🤦🏻‍♂️I mis-typed the identification string for your motherboard. Had MSI in front of the name by accident.

Thanks for the logs, that was an easy one to fix. It basically showed that your SIO controller was still being assigned as an NCT6687D instead of NCT6687D-R. So it was as if you were running the vanilla release of LHM... Sorry about that! Give this build a shot and report back.

[LHM_NCT6687DR_Build2512.zip](https://github.com/user-attachments/files/18447431/LHM_NCT6687DR_Build2512.zip)

Dr-Mugg, 2025-01-17:
Great find! It's working now but some of the temp sensors disappeared and voltages changed names. Maybe this is not part of this bug about rpm/control but I've collected whats missing and changed in this picture:

[image]

CPU Socket, M2_1 (not working) and PCIe x1 is missing. PCIe x1 is the T_SEN_1 external temp sensor for MSI probe. We could close this and start another issue since rpm/control is fixed.

closed 2025-01-17 by Alcolawl