Alcolawl/LibreHardwareMonitor #3: FanControl only Finds GPU fans after replacing LibreHardwareMonitorLib files

chinonso098, 2025-01-15:
**bug**
I am currently on v211 of FanControl(dotnet8 to be specific)

When i copy the LibreHardwareMonitorLib.* files from the Net472 folder into in my FanControl folder, then i delete the Configuration folder in FanControl to re-trigger fan detection and calibration, only the GPU fans are found.

If i restore the previous original LibreHardwareMonitorLib.* files that shipped with FanControl, and delete the Configuration
folder in FanControl to re-trigger fan detection and calibration, all fans are found.

**Screenshots**
With the LibreHardwareMonitorLib.* files from the Net472 folder

[image]

With the LibreHardwareMonitorLib.* files that shipped with FanControl

[image]

**Desktop (please complete the following information):**
 - OS: Win 11 23h2
- Motherboard: MSI MPG X870 Tomahawk WIFI

jukibom, 2025-01-16:
Make sure you're using the .NET 4.8 release of FanControl - the .NET 8 one won't work (I had to switch!)

Alcolawl, 2025-01-16:
Interesting... Works for me on my end with the .NET 8.0 version of FanControl. 

[image]

@chinonso098 Try @jukibom's suggestion just to test. I've tried to replicate your issue and I'm not able to do so. Would you mind attaching a log from FanControl by going to Settings, Edit sources..., and copying a log by clicking the copy button next to "LibreHardwareMonitor"?

[image]

This should give me an idea of whether or not the paths to the sensors and controls are being set properly.

chinonso098, 2025-01-17:
Ok, let me give that a whirl

chinonso098, 2025-01-17:
OK, here is what i got. I was able to get it (v211 FanControl .NET 4.8) to run. Using LibreHardwareMonitorLib.* files from the Net472

[image]

Alcolawl, 2025-01-17:
Nono, this is good news! Unfortunately you have to manually pair the speed sensors with the fan controls, click the "Pair Speed Sensor" for each fan you want to control. Remember that you can also show hidden cards by clicking the eye at the top right. That should get you going for now. 

I'll try to see if I can get .NET 6.0 and .NET 8 versions of LHM to work as well, but that's been a bit of a low priority thing for me at the moment and it doesn't quite behave right on my machine either.

chinonso098, 2025-01-17:
Thanks let me give that a whirl 😊

chinonso098, 2025-01-17:
Thanks, @Alcolawl.  FanControl is now operational 😊

Alcolawl, 2025-01-17:
Of course! Glad it's working for you now! :D

closed 2025-01-17 by Alcolawl