Alcolawl/LibreHardwareMonitor #4: Wrong voltages/temps MSI MPG Z890 Carbon Wifi

Dr-Mugg, 2025-01-17:
I've found that some voltages and temps are either missing or display wrong results when LHM lib-files are replaced. Theres some info the attached image but here's a short list, some of them are just cosmetic:

Voltages
- Vcore not measuring anymore: static @ 0,658 V but works in vanilla (this could be CPU Nortbridge/SoC in fork but there's some differences in the measured results.)
- DIMM wrong voltage: @ 0,77 V in fork, 1,3+ V in vanilla.
- CPU I/O changed name to Chipset: called CPU I/O in bios, no big deal.
- Vref different voltage: @1,262 V and @ 1,820 V

Temp
- CPU Socket missing
- PCIe x1 missing, this is T_SEN_1, MSI external temp probe
- M2_1 missing but fail to display any data even in original LHM

[image]

OS: Win11
Motherboard MSI MPG Z890 Carbon WIFI

Alcolawl, 2025-01-17:
This is going to seem redundant but you can actually attach a screenshot of LHM and HWinfo side by side? That should get this squared away pretty fast.

Dr-Mugg, 2025-01-17:
No problem, here you go:

[image]

Alcolawl, 2025-01-18:
Interesting. I think what's throwing me for a loop is not only did the voltage labels change, but so did the values. I've reset the values to nearly default while taking a stab at the right order, using HWInfo as a guide (If it's correct...), so thanks for that screenshot.

Temperature sensors should be correct now, it's like the X870E Carbon, which I suspected, but you get an extra CPU Socket temperature sensor that they don't. T_SEN_1 in Hwinfo is often incorrect on new motherboards and it's most likely the same thing here just like the X870E version. 

Try this new build and report back when you can.

[LHM_DrMugg_Build1.zip](https://github.com/user-attachments/files/18462313/LHM_DrMugg_Build1.zip)

Dr-Mugg, 2025-01-18:
Great! Temps are now correct and report the same values ​​across the board. Just to be sure I've even tested PCIe x1 (T_SEN_1) with MSI's temperature probe and compared it with an external thermometer which shows the same temp.
However, I can't even guess what VIN5? (Voltage 1 in LHM, VIN3 in Hwinfo) is supposed to represent, nothing in the bios resembles this.

About voltages:
- VIN4 now shows correct values and is DIMM/VDD2.
- Vcore is working, just some small delays between the applications in the screenshot.

[image]

I would say the issue is resolved as all temps are there and the known voltages.
Thank you for making this work.

Alcolawl, 2025-01-18:
Awesome! I'll make sure to update the VDIMM readout for the next release. Thanks for your help on this.

closed 2025-01-18 by Alcolawl