Alcolawl/LibreHardwareMonitor #5: [MOBO] Please, add Support for MSI MAG X870E TOMAHAWK WIFI (MS-7E59)

Siepnir, 2025-01-22:
First of all, thank you for all the work you did so far!

Would it be possible to add the support to the MOBO MSI MAG X870E TOMAHAWK WIFI (MS-7E59) ?
The controller should still be the same as the non-E variant of this mobo.

LHM correctly identifies the motherboard name, but the fan control % and speed RPM are wrong / nil.
I tried to set the fan to a fixed speed in the BIOS, to no effect.
I have 3 fan connectors in use, all by PWM fans:
- CPU
- Sys 2
- Sys 3

Let me know if you need any more infos!

Thank you!

[HWMonitor.txt](https://github.com/user-attachments/files/18509887/HWMonitor.txt)
[LibreHardwareMonitor.Report.txt](https://github.com/user-attachments/files/18509888/LibreHardwareMonitor.Report.txt)

Alcolawl, 2025-01-22:
Hey! So I've put together a quick build to get us started. This process may require a bit of back and forth, so please be patient, but we'll get it sorted out. :D

This new build should reveal a CPU Socket temperature, restore fan control and duty cycle readout, and shuffle up the voltage layout slightly. Please report back with screenshots, a comparison screenshot between this build of LHM and HWInfo, whether or not FanControl has truly been restored, and obviously any other weird behaviors you might come across.

To install, just unzip and take the contents of the Net472 folder and drop it in the same directory as LHM (obviously do this while LHM isn't running). Then run LHM again.
[LHM_Slepnir_Build1.zip](https://github.com/user-attachments/files/18510652/LHM_Slepnir_Build1.zip)

Siepnir, 2025-01-22:
Thank you for the swift reply (and fix!!!)
I attached the before/after screenshots of LHM and a comparison screenshot from HWMon.
The manual fan control seems to be working as well!

There's an issue somewhere when I try to use FanControl with the new build; I've attached the log file.
Thanks!

[LibreHardwareMonitor.Report_fixed.txt](https://github.com/user-attachments/files/18511369/LibreHardwareMonitor.Report_fixed.txt)
[LHM_screenshots.zip](https://github.com/user-attachments/files/18511330/LHM_screenshots.zip)
[fancontrol_log.txt](https://github.com/user-attachments/files/18511351/fancontrol_log.txt)

Siepnir, 2025-01-22:
Sure, here it is:
[image]

Even if I try to manually re-trigger the scan, FanControl doesn't find anything else, besides the GPU.

Yes, this is done in a fresh FanControl installation where I replaced all three files you provided.

Alcolawl, 2025-01-22:
Ok, please go to Settings > Edit sources... and click the copy button next to LibreHardwareMonitor. Then paste or attach the log here.

[image]

Siepnir, 2025-01-22:
Ok I might have found the reason why it's not working: I'm using the .NET 8.0 version of FanControl, which produces the following log:
[Log_FanControl_dotNet8p0.txt](https://github.com/user-attachments/files/18511925/Log_FanControl_dotNet8p0.txt)

Everything works fine when I switch to the .NET 4.8 version, my bad.

Alcolawl, 2025-01-22:
Yep, that's precisely it. You beat me to it, I was just typing up a quick reply to tell you to try the .NET 4.8 version of FanControl. My apologies, I need to remember to remind people of that when they open issues here and also put it in the README.

Glad you got it working, though, and thanks for helping me add support for another motherboard!

Siepnir, 2025-01-22:
Yeah sorry, I completely missed that!
Thanks to you for your incredibly swift fix, I'll keep an eye out for anything weird.

We can close this if you want, thank you!

closed 2025-01-22 by Alcolawl