Alcolawl/LibreHardwareMonitor #9: [MOBO] Add Support for MSI MAG B850 TOMAHAWK MAX WIFI (MS-7E62)

Maffia171, 2025-01-28:
Hi, I'm hoping to get support added for my motherboard as well. Didn't see any B850's in the issue list so far but I think I'm in the right spot with this chip.

Full name of Motherboard: MSI MAG B850 TOMAHAWK MAX WIFI (MS-7E62)
Nuvoton NCT6687D

Not sure if the HWMonitor log is also needed, I added both files:
[LibreHardwareMonitor.Report.txt](https://github.com/user-attachments/files/18580313/LibreHardwareMonitor.Report.txt)

[HWMonitor.txt](https://github.com/user-attachments/files/18580325/HWMonitor.txt)

Thanks in advance for any help you're able to provide!

Alcolawl, 2025-01-29:
Thanks for the logs! 

Give this build a shot and report back. This one should be pretty straight forward. Please read the README as it'll probably tackle most of the issues and oddities you might run into.

[LHM_NCT6687DR_Build2516.zip](https://github.com/user-attachments/files/18582386/LHM_NCT6687DR_Build2516.zip)

Maffia171, 2025-01-29:
Thanks for the new build files, it works great in both LHM and FanControl. 

[image]

[image]

Much appreciated that you're doing all this. I just got this board and it's great to not have to set these fan curves in the BIOS.

closed 2025-01-29 by Alcolawl