Alcolawl/LibreHardwareMonitor #10: Continued issues with x870 tomahawk on newest bios 7E51v1A24

WesFarley, 2025-01-30:
Hello! Me again :)
So I updated bios to the A24 version, and chipset drivers to 7.01.08.129, it promised bug fixes, apparently big mistake.  My windows credentials were corrupted and caused me to reinstall the OS and start from scratch, so that's pretty awesome.  Several hours later, once again trying to get fancontrol running on version v212, net 8 framework.  With that stock install I see the sensors I expect but 0 rpm displayed from those fan headers, also no response to trying to manually pair for rpms even with force control.  Once I slice in your forked files fancontrol just refuses to start.  One step forward = two steps back, sigh.  Any thoughts on the new situation?  

Still the same x870 mobo

Alcolawl, 2025-01-30:
Which .NET version of FanControl are you using? The .Net 4.8 seems to be the better bet. Give it a shot if you haven't already. Otherwise attach a screenshot of the error you get when attempting to start FanControl.

Also RIP. I know the pain of unexpectedly having to reinstall Windows. It's never fun.

WesFarley, 2025-01-30:
I was using the NET 8, installed the 4.8 NET and it seems to be working again with your fork files.   Sheesh, thank you!

It is never fun.  WinUtil is a great help to get started but its just a start.

closed 2025-01-30 by WesFarley