Alcolawl/LibreHardwareMonitor #11: [MOBO] Add Support for MSI X870 GAMING PLUS WIFI (MS-7E47)

soliduz1433, 2025-02-01:
MSI X870 GAMING PLUS WIFI (MS-7E47) with Nuvoto NCT6687D not showing fans in LHM

[LibreHardwareMonitor.Report.txt](https://github.com/user-attachments/files/18627485/LibreHardwareMonitor.Report.txt)

Let me know if there is anything else you need.

Much appreciated.

Alcolawl, 2025-02-03:
Thanks for the logs!

Give this build a shot and report back. This one should be pretty straight forward. Please read the README as it'll probably clear up most of the issues and oddities you might run into.

[LHM_Build_Soliduz1433.zip](https://github.com/user-attachments/files/18647369/LHM_Build_Soliduz1433.zip)

soliduz1433, 2025-02-03:
Looks like it works. Thanks!

closed 2025-02-03 by Alcolawl