Alcolawl/LibreHardwareMonitor #12: [MOBO] Add Support for MSI B850 GAMING PLUS WIFI (MS-7E56)

capri2k2, 2025-02-04:
Please add MSI B850 GAMING PLUS WIFI (MS-7E56) mainboard.
If you need any additional information, please let me know.

Thanks in advance for your support,
capri2k2

[LibreHardwareMonitor.Report_reduced.txt](https://github.com/user-attachments/files/18656848/LibreHardwareMonitor.Report_reduced.txt)

Alcolawl, 2025-02-05:
Thanks for the logs!

Give this build a shot and report back. This one should be pretty straight forward. Please read the README as it'll probably clear up most of the issues and oddities you might run into.

[LHM_Build_capri2k2.zip](https://github.com/user-attachments/files/18664828/LHM_Build_capri2k2.zip)

Alcolawl, 2025-02-05:
SYS_FAN1, 2, 3, 4 should all behave the same. Note that the control of the fan should work but the fan itself will respond quite slowly (about 1-2% per second) when first receiving the command. This is due to the way we're controlling the fans within the code and it's unfortunately the only way we've been able to get it to work. 

Please try setting the Control for SYS_FAN1 and see if it responds and spins up to full speed. It will take approximately 5-10 seconds. If the fan does not respond at all, report back.

capri2k2, 2025-02-07:
Hello,
first, I had to organize some more fans for testing.

The fan controls had been assigned wrong in Fancontrol.

After testing all 4 Sysfans, I can confirm that everything works like a charm.

Great job, many thanks!

I hope that in the future, there will be a solution that the Sysfans will be as responsive as the CPU fan.

Regards,
capri2k2

closed 2025-02-07 by Alcolawl