Alcolawl/LibreHardwareMonitor #13: Raising of PR

chrisgleissner, 2025-02-05:
Hello @Alcolawl ,

Fabtastic work on this. I read your entire investigation thread and truly appreciate all the time invested into this. I can confirm that it works great on an MSI 870 Pro motherboard and FanControl. This despite having the MSI Center installed as well.

There is currently a PR open against upstream by another author who mentions that fan control does jot work yet. It seems your PR is more advanced as it clearly does work on yours.

Are there any plans to get a PR raised for your fork so that it is merged into mainline? In case some workarounds may be objected by the maintainers of upstream, having a plugin for FanControl might be an alternative.

Many thanks
Christian

Alcolawl, 2025-02-05:
Hey Christian,

So, appreciate you taking the time to read through our struggles and how we got to this point. Also appreciate you reaching out. Your questions are straight forward but the answers are a bit more nuanced, so I'll try to be as clear as possible. 

First thing's first: I'm glad this fork is working well for you. When you only have a sample size of 1 for a motherboard, it can be tough to truly know if your solutions have worked, so it's always great to get confirmations from multiple different testers.

Now to your questions and observations:

- The current PR opened by @demomanca is based off of the work you see in this fork, but with a critical end-user component missing: System Fan Control. Him and I (as well as a few others) are working together to get support done for new MSI boards that use the Nuvoton NCT6687D Super I/O controller. However, we're basically stumped at the part where we exert full control over the system fans where we can get them to react, ramp up/down, etc as quickly as the CPU fan does. My solution contained within this fork is two-fold:
  - First, if your model motherboard matches a hard-coded list of motherboards (like the ones currently in the [README](https://github.com/Alcolawl/LibreHardwareMonitor/blob/master/README.md)) a variable that contains the Super IO Controller's ID is set to a new model that I created: NCT6687DR. Through the flow of the entire program, if the SIO controller variable is set to "NCT6687DR" it will get a different set of voltages, temperatures, fan speed and control register assignments, etc. This allows us to maintain compatibility with older motherboards that use the same NCT6687D SIO controller that don't exhibit the issues we have, while also cleanly adding support for newer motherboards that do. This also enables us to use the new fan control function I wrote...
  - The Fan Control function is necessary because we cannot find the registers responsible for exerting control over the system fans like the legacy code does for older motherboards. Each fan has a 7-point fan curve, of which each point is stored in a register. The solution here is to write all 7 points of the curve to the same value when a fan duty cycle is requested, this way the fan has no choice but to comply.
- Because of the above solutions are not perfect, I'm hesitant to create a PR to merge with LHM. As you've probably seen, this fork, while working, does not behave as normal users would expect it to and if it were to be merged with LHM in an update and bundled with FanControl, it would create a ton of headaches for that developer as well. With all of this in mind we thought it might be better that we generate a PR to at least get the voltage and temperature readouts correct and at least begin to support newer MSI motherboards, while maintaining a separate fork for AM5/LGA1851 MSI Motherboards until a better solution is found (if one ever is). 
- I know LHM and FanControl are technically two different programs, and perhaps a marge of this fork into upstream LHM is the right move, but I think we'd also have to coordinate with the creator of Fan Control to get the program to cooperate with our rather slow reacting fan control solution. It's something we've gone back and forth on, but it's also something that I've warmed up to as the list of supported motherboards continues to grow over the last few weeks.

Hope this provides some clarity. Let me know what you think.

Thanks,
Alcolawl

chrisgleissner, 2025-02-06:
Hi Alcolawl,

Thanks for your detailed reply. Your approach makes a lot of sense.

Considering that there may never be a solution for the slow fan reaction
time as it may be a hardware feature (even MSI's own control software seems
to only slowly adjust the system fan RPMs to temp changes), it may be worth
reaching out to the maintainer of FanControl and ask for a new feature:
"Configurable fan reaction time"

This way, a user of fan control could set a higher reaction time which is
then used by the auto calibration logic in that program.

It could also default to sensible values for specific fans that are slow to
react, such as seems to be the case for the system fans in the new MSI 870
motherboards.

I reached out to him before on a different topic and he was very helpful.

BTW, I did notice that the auto calibration in FanControl seems to work
after having manually paired fans with sensors. At least from a UI
perspective and from the behavior of the 4 system fans I have hooked up in
my MSI 870 Pro. They all ramp up and down just fine using FanControl and
your forked DLL, even after auto calibration.

As a workaround, for now, after a new release of FanControl, I can copy
your forked libs into the install folder each time. But so that your
efforts become more visible and usable by everyone, it would be great to
try and get things merged unto mainline instead of holding off until a
perfect solution is found.

Once again, I am very appreciative of all the efforts and ingenuity that
went into your fork and if makes my PC finally quiet again.

I only own this single MSI board, but I am a dev myself in Kotlin and Java,
and if I can help in any way, please let me know.

Kind regards
Christian

Alcolawl, 2025-02-06:
> "Configurable fan reaction time"
> This way, a user of fan control could set a higher reaction time which is then used by the auto calibration logic in that program. It could also default to sensible [...]

Yep, this is exactly what we discussed. I agree, I think it's time I reach out to Rem0o and see what he thinks of all this

> I did notice that the auto calibration in FanControl seems to work after having manually paired fans with sensors. At least from a UI perspective and from the behavior of the 4 system fans I have [...]
Since a picture is worth a thousand words, here's the curve points that were discovered by FanControl as part of its auto-calibration which I engaged after creating a new configuration and manually pairing fans to sensors as advised in your forked branch (thanks). So it does work, and I think it would work even better with the small (I think) change mentioned above.

This is exactly what I recommend users do in the README. It's still not perfect since the fans take _so long_ to ramp up that the wrong RPM is registered at the requested duty cycle. Most users don't notice this but it happens and requires manual calibration to smooth out, otherwise the program gets confused ("?" symbol next to the control).

I have family visiting this weekend, so I'll open an issue when I'm back at the keyboard next week and see if I or we can't get Rem0o to chat offline and figure something out.

Rem0o, 2025-02-06:
Hi,

Was just wondering if you were using https://github.com/Fred78290/nct6687d/blob/main/nct6687.c#L185  as a reference?
or https://github.com/torvalds/linux/blob/92514ef226f511f2ca1fb1b8752966097518edc0/drivers/hwmon/nct6683.c#L73 ?

Alcolawl, 2025-02-07:
> Hi,
> 
> Was just wondering if you were using https://github.com/Fred78290/nct6687d/blob/main/nct6687.c#L185 as a reference? or https://github.com/torvalds/linux/blob/92514ef226f511f2ca1fb1b8752966097518edc0/drivers/hwmon/nct6683.c#L73 ?
> 
> Couldn't help but notice this in [@Fred78290](https://github.com/Fred78290) version:
> 
> //#define NCT6687_FAN_CFG_DONE          0x40    //! for 6683 returns auto mode and clears 0xA00, 0xA28-0xA2F registers
> #define NCT6687_FAN_CFG_DONE            0x00    //! tested on 6683 6687
> 
> Where the "0xA00, 0xA28-0xA2F registers" would mean it clears the FAN_CONTROL_MODE_REG + FAN_PWM_COMMAND_REG, thus the mode and the PWM value.
> 
> Don't know if this could be one of the reasons the normal setting of a PWM value doesn't work.

Yeah that's been one of the biggest references during our research. Ironically, if you look at the comments, he wrote a lot of that based off of LibreHardwareMonitor code. The big problem here is that MSI seems to be using different registers than what were used previously on other NCT6687 boards. So the only solution I've found is explained in my extremely long comment above. We were considering approaching you and getting your thoughts on adding a "Slow Fan Reaction Mode" toggle or perhaps a configurable value to set for the automatic calibration, because the way it's currently configured the program is too "impatient" and fails to pair the speed sensors to the fan controls, which must be done manually. Curious what you think. I don't want to generate a PR with the fan control solution I've devised if it's going to negatively effect people downstream (i.e. People that use and love FanControl).

Rem0o, 2025-02-07:
@Alcolawl  

I've seen a similar case of slow ramp up/down with the CorsairLink plugin where auto-cal would fail for similar reason.

How long are we talking for the RPM value to stabilize ? Right now the auto-calibration will wait up to 20 seconds for any given % test.

Alcolawl, 2025-02-07:
@Rem0o They seem to respond at ~1-2% per second. It took about 37 seconds for my fans idling at 40% duty cycle to ramp up and stabilize at 100%.

Rem0o, 2025-02-07:
The initial ramp up to 100% is the longest theoretical wait.  Each step is then a drop of 10%, so I guess that would be much quicker?

Alcolawl, 2025-02-07:
I suppose the 60 second timeout could work, though it honestly might have to be a bit higher. I just tested this some more. It took my Arctic P14 just about 60 seconds to ramp up to 100% from 0%. Like I said earlier, the fans respond at <2% per second, so technically a run up to 100% from 0% could take even longer than a minute. But once the fans have stabilized at 100%, I clocked each 10% step down down to take less than 10 seconds, so the 20 second window there should still suffice.

I guess my concern was changing the timeouts or the general performance of your program for a niche case, which seems to be limited to new MSI AM5/LGA1851 socket motherboards, hence why I thought maybe a toggle option in Settings would be better. But then you run into an issue of normal users not being aware of the setting and then clogging up your git with issues about their fans not calibrating correctly, so it's a tough choice to weigh.

Rem0o, 2025-02-07:
Doesn’t need to be a setting.  It's a timeout value, so if the fan speeds stabilizes fast, it won't change anything for the user.

chrisgleissner, 2025-02-07:
I just tried the latest version 214 of FanControl with the changed DLLS of Alcolawl. I have Arctic P14 PWM case Fans as well and timed the spin up from 0 to 100 percent to be about a minute, as well. 

I had a DC fan (with 3 pins) from my Fractal Design case connected to the same fan header as a test, and even though the FanControl automatic calibration function worked reasonably well with the PWM fans (ie. it was able to create a few curve points as shown in the screenshot in my last comment in this thread), in DC mode, the reaction time was a bit slower and FanControl gave up without creating curve points.

The idea with the increased timeout value in FanControl sounds very good, since users with faster Fan reaction times won't be negatively affected, and those who have slower fans don't need to manually configure a specific fan wait time. Really looking forward as FanControl has become an invaluable asset in my system.

Rem0o, 2025-02-07:
Keep in mind also that calibration is optional.  It is only used if you create RPM mode fan curves.  If your fan curves are set in percentage, calibration is not used.

chrisgleissner, 2025-02-07:
If you need any beta testers for the timeout change discussed above, e.g.based on a FanControl fork or branch, please let me know.

Rem0o, 2025-02-07:
I did a few things to improve the assisted setup steps to allow for slower ramping fans without compromising speed when it is possible.  

I had a few hard coded wait times for the fan/speed pairing utility and calibration, I replaced them with a wait condition for rpm value to be stable + a long timeout.  If the value is stable => continue, else wait for long timeout.  This should be good.

Here is a preliminary version: 
[slow_ramping_fix.zip](https://github.com/user-attachments/files/18710152/slow_ramping_fix.zip)

chrisgleissner, 2025-02-07:
That's great news! I will try the auto calibration first thing tomorrow.

chrisgleissner, 2025-02-08:
@Rem0o @Alcolawl I tried the FanControl "Slow ramping" fix together with the latest build of this forked LHM repo and I am happy to report that it works beautifully! Fantastic work.

My setup:
- MSI 870 Pro Motherboard, latest firmware.
- AMD 9950X CPU
- Windows 11 Pro 2024H2, latest updates.
- 4 case PWM fans, connected to 4 "system fan" headers. These are Arctic P14 fans.
- 2 CPU PWM fans from my Noctua DH15, connected to the CPU and pump headers.
- 2 GPU fans from my Asus Nvidia 3060 GPU. These are attached directly to the GPU.

Steps I took:
- Shut down the FanControl which I normally use and which starts with Windows.
- Unzipped the `slow-ramping_fix.zip` from the attachment in this thread from 18h ago into a new folder.
- Copied the files from this LHM repo's latest release (2521) into the same folder.
- Started FanControl from that folder.
- I then just went through the various wizard screens that FanControl shows:
  - Fan discovery: All worked. It discovered the 4 system fans connected to the MSI motherboard which previously were not discovered, using the latest "official" FanControl.
  - Fan autocalibration: All worked. It started with 100% and was able to quickly complete the CPU fan calibration. The fan calibration of the 4 system fans took longer, as expected due to the slow ramp issue.
- I then finally manually created various curves and can once more enjoy a silent PC with full FanControl and LHM support!

I attached screenshots of the entire installation flow, showing every change in the UI. (May have missed a few screenshots where it confirmed that the GPU was fully calibrated as these flashed by very quickly.

[FanControl_SlowRampingFix_LHM_Build2521.zip](https://github.com/user-attachments/files/18718732/FanControl_SlowRampingFix_LHM_Build2521.zip)

A lot of work went into this, so I'd just like to say thank you very much. Now I just hope that it'll be smooth sailing to get this LHM fork merged into upstream, and then integrated into an official new FanControl version. :-)

mbonca, 2025-02-08:
My FanControl only fines 2 fans of GPU. No others. Do I have to set them to manual (or something) in the BIOS?

chrisgleissner, 2025-02-08:
@mbonca Did you use the steps described in my last post to setup FanControl? I had the issue thst you described when I used the official FanControl version. But all issues were resolved with the patched FanControl attached to thumis thread, and installing the patched Li rrHardwareMonitor release into it.

mbonca, 2025-02-08:
@chrisgleissner I copied and overritten those 3 dll files into the FanControl folder, but as said, I only se the fans of GPU, none other.
I think I remember guys were saying if fans are controled from the Bios, you have to set them to manual (or something) in there, in order to see and set them inside FanControl app (with this none official version). I just wanna get some confirmation before I go my self back into the bios changing things.

chrisgleissner, 2025-02-08:
In my case, I recently reset my MSI bios to defaults using the Cmos reset
button at the motherboard rear.

I am not saying that this is needed, but in my case, it helped with WiFi
issues where Windows 11 Pro would not discover the Realtek WiFi of the MSI
870 Pro despite having installed the drivers from the MSI website. If you
did not make lots of customization in the Bios, you have little to lose
when trying this.

Also, just to double check, did you start FanControl by double clicking on
the exe that is in the Zip file on this thread?

mbonca, 2025-02-08:
> Also, just to double check, did you start FanControl by double clicking on
> the exe that is in the Zip file on this thread?

@chrisgleissner no, I didn't. Do you mean this file: [slow_ramping_fix.zip](https://github.com/user-attachments/files/18710152/slow_ramping_fix.zip)

I only downloaded those 3 files, and pasted them into FanControl folder.

mbonca, 2025-02-08:
Wow, seems that its working :) Thanks a lot @chrisgleissner

mbonca, 2025-02-08:
Wow, seems that its working :) Thanks a lot @chrisgleissner 

[image]

chrisgleissner, 2025-02-09:
That's great @mbonca . I am glad it works for you, as well. Thanks for confirming with the screenshot.

bitblair, 2025-02-11:
Hi, I tried this as well on a MSI MPG X870e Carbon Wifi with latest bios (JAN-20-25). The LHM fix works for me after following readme instructions and manually calibrating.

However, the slow ramping fix only marginally improved auto detection/calibration. I have 1-3 SYS_FAN populated and without the fix, none of these are auto-detected. With the fix SYS_FAN 1 is detected (Liquid Freezer III VRM Fan) but my case fans are not. These are stock Antec fans that ship with Flux Pro. I can see the RPM slowly increase during detection, it's around 2% a tick. But the auto detection gives up after around 10-12 seconds.

I can follow up later with logs if requested. Happy to help troubleshoot.

chrisgleissner, 2025-02-11:
Hi, just to double check, are the non-responsive fans connected as PWM fans
(4 pins) or DC fans (3 pins)? Also, do you have any other fan control
software running at the same time, such as MSI Center? It may also be good
to know how you configured these system fans in the Bios, i.e. whether you
used Auto, PWM, or DC. Thanks.

bitblair, 2025-02-11:
SYS Fans are all 4 pin, although most of them are connected to mobo via fan controller (SYS_FAN 1 is not).

bios config:

CPU Fan 1 - Smart Fan Mode enabled, Fan Type Auto Detect enabled, PWM
Pump Fan 1 - Smart Fan Mode disabled, Fan Type Auto Detect enabled, PWM
SYS Fan 1 - Smart Fan Mode disabled, Fan Type Auto Detect enabled, PWM
SYS Fan 2 - Smart Fan Mode disabled, Fan Type Auto Detect enabled, PWM
SYS Fan 3 - Smart Fan Mode disabled, Fan Type Auto Detect enabled, PWM

I have paired them manually and also manually calibrated based on README instructions in this repo. They respond well to graphs in fan control, just take a little while to spin up, down (to a point where the RPMs have a ? in Fan Control sometimes, but it usually corrects itself after a few seconds).

 I could look into plugging some of these directly into mobo to see if that makes a difference. I will report back later on this.

Rem0o, 2025-02-11:
> With the fix SYS_FAN 1 is detected (Liquid Freezer III VRM Fan) but my case fans are not. 

At which step EXACTLY does it "not get detected" ?

Fan pairing now waits up to 60 seconds.
Fan calibration waits also over a minute to get to 100%.

bitblair, 2025-02-11:
> At which step EXACTLY does it "not get detected" ?
> 
> Fan pairing now waits up to 60 seconds. Fan calibration waits also over a minute to get to 100%.

It was during speed pairing step. It could not detect pairing for SYS_FAN 2,3. I needed to manually pair speeds for them.

I still suspect either the fans or external fan controller is at fault here because the fix _did_ work for another fan connected directly to the mobo. But it could be coincidence. Like I said in my previous post, I'll try and isolate it probably connect one of the fans to the mobo to see if anything changes. Will report back later today.

Alcolawl, 2025-02-12:
Apologies for my late response, @Rem0o, I had family in town for a long weekend SuperBowl visit type thing and didn't have a change to respond until they left.

Good News and Bad News:

- Bad News first. When creating a new configuration, as if someone has installed and run the program for the first time, automatic pairing of the speed sensors to the system fan controls still fails. The fans simply do not respond fast enough, so I have a feeling that timeouts may need to be expanded there as well.
- Good News is, like many others have reported, manually pairing the speed sensors to the controls and then running the calibration wizard works great!

I've recorded a whole video of the process if more context is needed. Just trying to avoid making the user experience of your program worse. Thanks for all of the help so far.

bitblair, 2025-02-12:
> * Good News is, like many others have reported, manually pairing the speed sensors to the controls and then running the calibration wizard works great!
>
> I've recorded a whole video of the process if [...]

Wanted to follow up to say this is my experience as well. Still fails on my SYS_FAN 2,3 for initial speed pairing, but after pairing manually I can recalibrate paired control automatically. Satisfies my needs for now.

chrisgleissner, 2025-02-12:
Interesting. So maybe I was lucky? All my fans are Arctic P14 PWM, hooked directly into the system fan connectors of my MSI 870 P, latest Bios, Cmos reset to defaults. Pairing worked without any glitch. Plus I think at least one other person here shared my experience.

As mentioned, maybe extending the timeout even more will help. As I cannot think of another explanation, given your fans respond to manual changes in FanControl.

Alcolawl, 2025-02-12:
During automatic pairing, all controls are set to 75%. The CPU Fan pairs nearly immediately, as expected, my other populated fan headers PUMP (Which I expect to fail, the pump has no RPM out on my Arctic LF2 for whatever reason), and System Fan 1 and 2 all timeout at around 15 seconds and it moves on to the next fan.

bitblair, 2025-02-12:
> The timeout is set to over a minute for pairing each fan. Do you get that long of a try before it fails?

For me it only spent around 12 seconds before it indicated it couldn't detect.

Rem0o, 2025-02-12:
[Slow_ramping_fix_v2.zip](https://github.com/user-attachments/files/18769724/Slow_ramping_fix_v2.zip)

Forgot to change one timeout for when the pairing utility waits for a RPM drop.   Made it a bit smarter, first waiting for "any" rpm drop for 15 seconds, then waits for the full RPM drop for another 15 seconds *IF* there was at least a small RPM drop.

mbonca, 2025-02-12:
@Rem0o so we install the V2 now? Do we still need to do any manual pairing like in the previous version, or won't be necessary?

Rem0o, 2025-02-12:
@mbonca  if you have it working already, don't do anything.

mbonca, 2025-02-12:
Ok, it seems it does work quite fine for me indeed.

chrisgleissner, 2025-02-12:
I tried the `Slow_ramping_fix_v2.zip` attached above. As before, I unzipped it, then applied the latest release of this forked LHM branch on top, then started it. As with V1, the fan discovery and automatic calibration on my MSI Pro 870-P with 4 Arctic P14 PWM system fans worked flawlessly.

Here's a video of the entire flow, from start of the application to completion of the auto-calibration, including some scrolling through the generated fan curves: https://www.youtube.com/watch?v=zgS-h4Py7iA

Thanks a lot for the patch, and I hope this can be merged into mainline soon.

Rem0o, 2025-02-12:
@chrisgleissner seems to go pretty much as fast as it can, with great calibration curves. This is will in a near-future release.

Alcolawl, 2025-02-12:
> [Slow_ramping_fix_v2.zip](https://github.com/user-attachments/files/18769724/Slow_ramping_fix_v2.zip)
>
> Forgot to change one timeout for when the pairing utility waits for a RPM drop. Made it a bit [...]

Happy to report this works perfectly!

I will work to get my repo in a better state and open a Pull Request to have my changes merged into LHM... See what they say.

Alcolawl, 2025-02-14:
Pull Request submitted! Thanks for all the help thus far, everyone.
https://github.com/LibreHardwareMonitor/LibreHardwareMonitor/pull/1644

Rem0o, 2025-02-19:
Yeah just stay on your current version, or do the dll dance after an update if you absolutely want the update.

mbonca, 2025-02-20:
The FanControl app suggested me to update to v215, and so I did. And lost all my calibrations, which I though it might happen. Then I manually calibrated fans with the appropriate sensors, and it's not working again, the rpms are not shown, although fans are spinning.
Will have to go back to v214, the cracked one version I guess. How are your experences with v215?

Zanskie, 2025-03-05:
Hello,

im a new user of FanControl.

Today i updated the latest version of BIOS 7E47v1A34 for my MSI PRO X870-P WIFI, it seems the case fans #2 and #3 can be detected but the software unable to control the speed of the fans. no matter how i do it it still on default. Weirdly i tested on MSI Center also still the same.

CPU Radiator fans and AIO Pump are fine

Im using the latest version of FanControl + LHM Build 2522.

does anyone else with the same board having the same issue?

Thanks

DOOGLAK, 2025-03-09:
> Hello,
> 
> im a new user of FanControl.
> 
> Today i updated the latest version of BIOS 7E47v1A34 for my MSI PRO X870-P WIFI, it seems the case fans 2 and 3 can be detected but the software unable to control the speed of the fans. no matter how i do it it still on default low speed. Weirdly i tested on MSI Center also still the same.
> 
> CPU Radiator fans and AIO Pump are fine
> 
> Im using the latest version of FanControl + LHM Build 2522.
> 
> Tried [Slow_ramping_fix_v2.zip](https://github.com/user-attachments/files/18769724/Slow_ramping_fix_v2.zip) and v1 and same problem
> 
> does anyone else with the same board having the same issue?
> 
> Thanks

I'm having the same issue, no luck with either version - MSI X870E Tomahawk Mobo, newest drivers/chipset as of today.

Zanskie, 2025-03-10:
> I'm having the same issue, no luck with either version - MSI X870E Tomahawk Mobo, newest drivers/chipset as of today.

glad to hear that im not the only one with the problem. but anyway i opened a case with MSI and i think they just confirmed it and they will release new version to fix this issue.

[image]

mbonca, 2025-03-12:
System fans from 1 to 6 used to work, but not anymore from today on. Anyone knows why? I am using Fix_v2. Tried V1 and V2 agyin, and the same.

[image]

chrisgleissner, 2025-03-14:
Thanks for reaching out to MSI about this @Zanskie . I have not yet tried the latest BIOS 7E47v1A34 and I am still on 7E47v1A3 on my PRO X870-P WIFI where the fan auto discovery and subsequent control work great, using the patched v3 of FanControl (published higher in this thread, and since then also merged back into its official released version), together with the LibreHardwareMonitor release published in this fork.

Based on your experience, it sounds like a good idea not to 7E47v1A34 until MSI releases a fix.

Zanskie, 2025-03-15:
> Thanks for reaching out to MSI about this [@Zanskie](https://github.com/Zanskie) . Do you happen to have a link to this exchange with MSI so others can track any progress, or is this a private case you raised with them?
> 
> I googled this issue and other users seem to be affected, see https://forum-en.msi.com/index.php?threads/with-the-new-bios-7e47v1a34-for-pro-x870-p-wifi-now-fans-are-static-at-about-60.410151/
> 
> From that thread: "Same issue here. After updating to BIOS 7E47v1A34 on my PRO X870-P WIFI, the fans (both system and CPU) were locked to a fixed ~60% of the RPM. Even if I set full speed on bios, the fans still spin at ~60%. Reverting the BIOS solved the issue."
> 
> I have not yet tried the latest BIOS 7E47v1A34 and I am still on 7E47v1A3 on my PRO X870-P WIFI where the fan auto discovery and subsequent control work great, using the patched v3 of FanControl (published higher in this thread, and since then also merged back into its official released version), together with the LibreHardwareMonitor release published in this fork.
> 
> Based on your experience, it sounds like a good idea not to upgrade to firmware 7E47v1A34 until MSI releases a fix. If you are already on this version of the firmware but do not need its features (see https://www.msi.com/Motherboard/PRO-X870-P-WIFI/support) and rather have working fan control, it's also an idea to downgrade to version 7E47v1A3.

Well yes, i opened standard support case to MSI about this problem. About 2 days ago MSI actually released a new version E7E47AMSI.1A36 for the MSI PRO X870-P and now the fans speed control works properly. they gave me a google link to download directly though. I just checked they still haven't upload it to the board's main support page.

i think issue is with MSI, FanControl and LHM has nothing to do with it. As far i can see, its only matter of adding support to the newer chipset.

closed 2025-03-20 by Alcolawl