Alcolawl/LibreHardwareMonitor #14: FanControl won't launch after PC restart

norekluk, 2025-02-06:
**Describe the bug**
FanControl won't launch after PC restart. I've applied your solution, set up my fans and was happy. Then I turned my computer off. After turning it on again FanControl wouldn't launch. I tried running it as administrator. 

**To Reproduce**
Steps to reproduce the behavior, if applicable:
1. Apply your solution
2.Set up fans
3. Set launch on startup
4. Turn off computer
5. Turn on computer
6. FanControl won't launch

**Expected behavior**
I expected FanControl to start

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - OS: Windows 11 Pro
- Motherboard: MAG X870 TOMAHAWK WIFI

**Additional context**
Add any other context about the problem here.

Alcolawl, 2025-02-07:
Odd. Gotta admit this is a new one for me, no one else seems to have run across this issue yet. 
Mind attaching a log from FanControl? Navigate to Settings > Edit Sources... And click the clipboard button in the screenshot.

[image]

norekluk, 2025-02-07:
Well, I had to reinstall and apply your patch again even to be able to launch FanConrol. Is there a way how to get a log without having the FanControl opened? 

This is a log from right now. I haven't turned off my computer since reinstalling.

LibreHardwareMonitor Report

--------------------------------------------------------------------------------

Version: 0.9.4.0

--------------------------------------------------------------------------------

Common Language Runtime: 4.0.30319.42000
Operating System: Microsoft Windows NT 6.2.9200.0
Process Type: 64-Bit

--------------------------------------------------------------------------------

Sensors

|
+- MSI MAG X870 TOMAHAWK WIFI (MS-7E51) (/motherboard)
|  |
|  +- Nuvoton NCT6687D-R (/lpc/nct6687dr/0)
|  |  +- +12V           :   12.216   12.144   12.288 (/lpc/nct6687dr/0/voltage/0)
|  |  +- +5V            :     5.03     5.02     5.05 (/lpc/nct6687dr/0/voltage/1)
|  |  +- CPU Northbridge/SoC :     1.03     1.03    1.038 (/lpc/nct6687dr/0/voltage/2)
|  |  +- DIMM           :     1.12    1.116    1.124 (/lpc/nct6687dr/0/voltage/3)
|  |  +- Vcore          :    1.082     0.94    1.194 (/lpc/nct6687dr/0/voltage/4)
|  |  +- Chipset        :    0.312    0.288     0.34 (/lpc/nct6687dr/0/voltage/5)
|  |  +- CPU System Agent :     0.78    0.776    0.906 (/lpc/nct6687dr/0/voltage/6)
|  |  +- +3.3V          :    3.332     3.32    3.344 (/lpc/nct6687dr/0/voltage/8)
|  |  +- VREF           :    2.044        0    2.044 (/lpc/nct6687dr/0/voltage/9)
|  |  +- +1.8V          :    1.836        0    1.844 (/lpc/nct6687dr/0/voltage/10)
|  |  +- +3V Standby    :    3.348        0    3.356 (/lpc/nct6687dr/0/voltage/11)
|  |  +- AVSB           :    3.348        0    3.356 (/lpc/nct6687dr/0/voltage/12)
|  |  +- CMOS Battery   :    3.124        0    3.128 (/lpc/nct6687dr/0/voltage/13)
|  |  +- CPU Core       :       40       34       77 (/lpc/nct6687dr/0/temperature/0)
|  |  +- System         :     43.5       34     44.5 (/lpc/nct6687dr/0/temperature/1)
|  |  +- VRM MOS        :     43.5     30.5     46.5 (/lpc/nct6687dr/0/temperature/2)
|  |  +- Chipset        :       38     30.5     38.5 (/lpc/nct6687dr/0/temperature/3)
|  |  +- CPU Fan        :      383      298     1263 (/lpc/nct6687dr/0/fan/0)
|  |  +- Pump Fan #1    :        0        0        0 (/lpc/nct6687dr/0/fan/1)
|  |  +- System Fan #1  :     1275      597     2164 (/lpc/nct6687dr/0/fan/2)
|  |  +- System Fan #2  :     1227      531     2094 (/lpc/nct6687dr/0/fan/3)
|  |  +- System Fan #3  :      593      539     1071 (/lpc/nct6687dr/0/fan/4)
|  |  +- System Fan #4  :        0        0        0 (/lpc/nct6687dr/0/fan/5)
|  |  +- System Fan #5  :        0        0        0 (/lpc/nct6687dr/0/fan/6)
|  |  +- System Fan #6  :        0        0        0 (/lpc/nct6687dr/0/fan/7)
|  |  +- CPU Fan        :       17        5       70 (/lpc/nct6687dr/0/control/0)
|  |  +- Pump Fan       :       70       51       89 (/lpc/nct6687dr/0/control/1)
|  |  +- System Fan #1  :       22        6       50 (/lpc/nct6687dr/0/control/2)
|  |  +- System Fan #2  :       22        6       50 (/lpc/nct6687dr/0/control/3)
|  |  +- System Fan #3  :       22        6       50 (/lpc/nct6687dr/0/control/4)
|  |  +- System Fan #4  :       50       50       81 (/lpc/nct6687dr/0/control/5)
|  |  +- System Fan #5  :       50       50       81 (/lpc/nct6687dr/0/control/6)
|  |  +- System Fan #6  :       50       50       81 (/lpc/nct6687dr/0/control/7)
|
+- AMD Ryzen 7 9800X3D (/amdcpu/0)
|  +- Core #1 VID    :  0.49375  0.38125    0.575 (/amdcpu/0/voltage/2)
|  +- Core #2 VID    :  0.49375  0.38125  0.55625 (/amdcpu/0/voltage/3)
|  +- Core #3 VID    :  0.49375  0.38125  0.55625 (/amdcpu/0/voltage/4)
|  +- Core #4 VID    :  0.49375  0.38125  0.55625 (/amdcpu/0/voltage/5)
|  +- Core #5 VID    :  0.49375  0.38125  0.55625 (/amdcpu/0/voltage/6)
|  +- Core #6 VID    :  0.49375  0.38125  0.55625 (/amdcpu/0/voltage/7)
|  +- Core #7 VID    :  0.49375  0.38125  0.55625 (/amdcpu/0/voltage/8)
|  +- Core #8 VID    :  0.49375  0.38125  0.55625 (/amdcpu/0/voltage/9)
|  +- Package        :  26.2058  15.5763  110.444 (/amdcpu/0/power/0)
|  +- Core #1 (SMU)  :  1.08157 0.0311767  11.1657 (/amdcpu/0/power/1)
|  +- Core #2 (SMU)  : 0.896506 0.0223152  11.3108 (/amdcpu/0/power/2)
[... 1712 lines cut ...]
Ryzen SMU

 PM table version: 0x00620105
 PM table supported: True
 PM table layout defined: False
 PM table size: 0x000
 PM table start address: 0x3EE80000

 PM table dump:
  Idx    Offset   Value
 ------------------------
    0    0x000    0
--------------------------------------------------------------------------------

Intel GPU (D3D)

Number of adapters: 2

AdapterIndex: 0
DeviceId: \\?\PCI#VEN_10DE&DEV_2705&SUBSYS_41361458&REV_A1#4&2af8c5c3&0&0009#{1ca05180-a699-450a-9a0c-de4fbe3ddd89}
IsIntel: False

AdapterIndex: 1
DeviceId: \\?\ROOT#BasicRender#0000#{1ca05180-a699-450a-9a0c-de4fbe3ddd89}
IsIntel: False

chrisgleissner, 2025-02-08:
I have been using this forked LHM on the latest FanControl for the past 3 days. I am also using Windows 11 Pro (recent new install, 2024H2) on an MSI 870 Pro. The startup issues don't seem to be related to LHM, which is the only thing that was changed by this fork.

If you start FanControl manually, does it start or crash? If it starts, there could be an issue in how it was registered to be launched during Windows boot. 

I suggest to uninstall FanControl, check all traces of it are gone (verifying that it was removed from the file system should suffice), then reinstall the official latest version of FanControl and do not yet apply this LHM fork. Please verify whether restarting your machine several times now correctly restarts FanControl each time. If that works, you could then proceed to copying  the DLL and other files of the latest release of this LHM fork into your FanControl folder afte4 taking a backup of it. Does FanControl not start anymore after doing so?

I am just a user like you, not a developer of LHM. But the two components are separate and FanControl uses LHM, not the other way around. The only thing I can think of which could explain your behavior is if FanControl tried to use LHM, but this failed, and FanControl then crashed. If the above steps (basically a clean reinstall of the software) don't resolve your issue, it may be worth checking if FanControl leaves any traces of a startup attempt behind, e.g. Log file or Windows event log entry.

norekluk, 2025-02-08:
No, it didn't start, even manually. It blinked on the bar, and I could briefly see it in the task manager, but it wouldn't launch. I will try your testing approach. Thanks.

Alcolawl, 2025-03-20:
I cannot replicate this issue, and suspect it is either BIOS related, .NET related,  or LHM/FanControl are being interfered with by another monitoring program. I've had many weird behaviors occur when running multiple hardware monitoring programs alongside LHM bundled into FanControl including the complete inability to control the system fans or read temperatures accurately.

closed 2025-03-20 by Alcolawl