Alcolawl/LibreHardwareMonitor #15: [MOBO] Add Support for MSI MEG X870E GODLIKE (MS-7E48)

Michipulatos, 2025-02-06:
MSI MEG X870E GODLIKE (MS-7E48) with Nuvoto NCT6687D not showing fans in LHM

[LibreHardwareMonitor.Report.txt](https://github.com/user-attachments/files/18698950/LibreHardwareMonitor.Report.txt)

Happy to test and provide help as needed, thanks!

Alcolawl, 2025-02-07:
Thanks for the logs!

Give this build a shot and report back. Please read the README as it'll probably tackle most of the issues and oddities you might run into.

[LHM_Build_Michipulatos.zip](https://github.com/user-attachments/files/18712727/LHM_Build_Michipulatos.zip)

3dRikal, 2025-02-10:
Also works on MSI X870 GAMING PLUS WIFI.

Alcolawl, 2025-02-12:
> This seems to work pretty well! Some initial issues with detection and calibration but after some tweaking on my, everything is g2g. Thank you so much

Awesome, thanks for the help with testing! As for the detection and calibration issues in FanControl, we're currently working with the author of that program to get those sorted out. Testing has gone well so far. Look for an update in the future if and when this repo's improvements are merged into the upstream LHM master branch.

closed 2025-02-12 by Alcolawl