Alcolawl/LibreHardwareMonitor #16: [MOBO] Add Support for MSI B850 Gaming Plus WiFi

mjouve-jul, 2025-02-08:
Hello,

First time ever posting an issue on GitHub so my apologies if everything is posted as expected. 

I think I have searched everywhere in this repo and couldn't find anything related to the support of MSI B850 Gaming Plus WiFi. 

Like a lot of users recently, SYS_FANS are not showing neither in Fan Control or LHM, though they are detected by HW Monitor. See screenshot below for proof : 

[image]

I made sure to follow instructions first and copy the 3 files (dll, pdb, xml) to the LHM & Fan control folders and then restart Fan Control and create a new profile but it didn't work, unfortunately.

Also attached, LHM report as well as the HW Monitor one :

[LibreHardwareMonitor.Report.txt](https://github.com/user-attachments/files/18718973/LibreHardwareMonitor.Report.txt)

[HWMonitor.txt](https://github.com/user-attachments/files/18718981/HWMonitor.txt)

I see you are very pro-active on these issues and I thank you very much for that, your hard work is greatly appreciated.

Cheers

Alcolawl, 2025-02-12:
This is a duplicate issue, I'm sure you just didn't see it, so no harm no foul. You can check out the test build in that issue or you can wait for the next release coming tonight, which adds support.

closed 2025-02-12 by Alcolawl

mjouve-jul, 2025-02-12:
@Alcolawl I surely didn't see it, thank you very much, I just pasted the files in Fan control and it's able to detect my SYS_FANs and their RPMs, thank you very much!!

I will work on setting everything at a later time.

Thank you for all your efforts !