Alcolawl/LibreHardwareMonitor #17: [MOBO] Add Support for MSI MAG Z890 TOMAHAWK WIFI (MS-7E32)

tdpeek3, 2025-02-11:
Hello - I'm opening an issue to add support for the following motherboard:

**MSI MAG Z890 TOMAHAWK WIFI (MS-7E32)**

I've attached the hardware monitor registries per your request. 

[hardware_monitor_registries.txt](https://github.com/user-attachments/files/18745447/hardware_monitor_registries.txt)

Your fork worked perfectly with both identifying/supporting the fan's values in LHM. Thanks for all of your work!

[image]

Let me know if I can assist in any other way.

Alcolawl, 2025-02-12:
Awesome, thanks for confirming those fan sensors are correct. I created a test build for someone else and they never got back to me. Would you mind testing a quick build and confirming the voltage and temperature sensors look to be correct to the best of your knowledge? Perhaps a screenshot comparing next to HWInfo. Looks like the DIMM voltage readout is incorrect in that picture (Most likely just being cut in half).

[LHM_Build_Beta2522.zip](https://github.com/user-attachments/files/18761345/LHM_Build_Beta2522.zip)

tdpeek3, 2025-02-12:
Hey, no problem! Let me know if you need any other screenshots.

[image]

Alcolawl, 2025-02-12:
That actually looks pretty good! I'll rename Unknown_3 temperature sensor to VIN7 but everything else looks accurate. Thanks for the quick response!

tdpeek3, 2025-02-12:
You're the one I should be thanking! :) I appreciate you spending your free time doing this.

Alcolawl, 2025-02-12:
Of course!

closed 2025-02-12 by Alcolawl