Alcolawl/LibreHardwareMonitor #18: Add support for MSI MEG Z890 ACE (MS-7E22)

Falerion-git, 2025-02-11:
Hello,

I'm opening this issue to provide the data for the

MSI MEG Z890 ACE (MS-7E22) mainboard.

Here ist the "Hardware Monitor Register" part of LibreHardwareMonitor:

[LibreHardwareMonitor.Report-Registers.txt](https://github.com/user-attachments/files/18758222/LibreHardwareMonitor.Report-Registers.txt)

Happy to test any new versions.

Alcolawl, 2025-02-12:
Thanks for the log, unfortunately the info I had in the README was wrong, and while you produced a report that followed the directions correctly, it basically had no useful info in it in terms of making sure sensors were getting read correctly. That's totally my fault, so I apologize!

Luckily, I was already working on a test build for someone else a took a stab at adding support for your motherboard too. Please report back with any incorrect readings, missing sensors, etc. Also, please read the README as it'll probably clear up most of the issues and oddities you might run into. Thanks in advance!

[LHM_Build_Beta2522.zip](https://github.com/user-attachments/files/18761395/LHM_Build_Beta2522.zip)

Falerion-git, 2025-02-12:
Sure, no problem.
I tested the beta 2522 and it looks like that the fan speeds are read now.
But I'm, not sure about the controls yet.

[image]

Would it help to upload the whole report? Or the report from the beta 2522 version?

Alcolawl, 2025-02-12:
Well, it looks like the voltage and temperature sensors are correct, so that's good news. 

Controls should be easy to test. Try setting the controls of your system fans to 100% and see if they respond. A report shouldn't be necessary at this point.

closed 2025-03-20 by Alcolawl