Alcolawl/LibreHardwareMonitor #19: [MOBO] Add Support for MSI PRO B850-P WIFI (MS-7E56)

DoinMyBest777, 2025-02-12:
Could you add MSI PRO B850-P WIFI (MS-7E56) to the list please? Fans System 1 to 4 are not detected, only CPU Fan and GPUs are detected (i don't use the "Pump" header so I don't know about this one.) 

Here's the register part(?):

[LibreHardwareMonitor.Report.txt](https://github.com/user-attachments/files/18761638/LibreHardwareMonitor.Report.txt)

Thanks

Alcolawl, 2025-02-12:
Thanks for the logs!

Give this build a shot and report back. Please read the README as it'll probably tackle most of the issues and oddities you might run into.

[LHM_Build_Beta2522_2.zip](https://github.com/user-attachments/files/18761689/LHM_Build_Beta2522_2.zip)

DoinMyBest777, 2025-02-12:
[image]
Managed to take control of the 4 fans header. The 4th still show nothing though...  But it's probably because it's connected to a hub. I can still control the their speeds so I'm good
Anyway, big thanks!

closed 2025-02-12 by Alcolawl