Alcolawl/LibreHardwareMonitor #20: Add support for MSI Pro Z890-A WiFi (MS-7E32)

Snakeoilsnaker, 2025-02-16:
Hello,

I am providing a LHM Report for the MSI Pro Z890-A WiFi. It seems this is just a relabeled Z890 Tomahawk. 

I tested 2522 with latest portable and .net version of FanControl also with latest LHM but System Fans seem to not report back any RPM.

I am curious does every MSI board handle the Nuvoton NCT6687D differently?

Thank you for all of your work. Your efforts are greatly appreciated

all the best

[LibreHardwareMonitor.Report.txt](https://github.com/user-attachments/files/18815288/LibreHardwareMonitor.Report.txt)

Alcolawl, 2025-02-16:
Yeah, MSI modified the NCT6687D in some way, so the existing code doesn't really work for the newer MSI Motherboards. Check out the PR for merging my changes into the LHM Master branch for a bit more context if you'd like: https://github.com/LibreHardwareMonitor/LibreHardwareMonitor/pull/1644

Give this build a shot and report back. Please read the README as it'll probably tackle most of the issues and oddities you might run into.

[LHM_Build_Snakeoilsnaker.zip](https://github.com/user-attachments/files/18816074/LHM_Build_Snakeoilsnaker.zip)

Snakeoilsnaker, 2025-02-17:
Hey Alcolawl,

thank you that was an interesting read! Didn't expect such a fast response. 

I've tested your patch and Fan Control is now working with the MSI Pro Z890-A. LHM is reporting Fanspeeds correctly now and do match with MSI BIOS fan speeds. 

There seems to be an issue with MSI implementation of the controller even at the lowest BIOS setting for fan speed up/down-spin time the Fans seem to not be recognized by Fan Controls automated setup and will skip with "No Speedsensor reacted". Installed fans just take too long to react to fan controls request to change RPM. 

As for now manual setup is required but who knows maybe a fix will  be implemented by MSI (I will report that bug to MSI) or Fan Control can implement an option to wait at least some user specified time before going to the next controller. 

One Fan array (3x Noctua AF-12) needs a manual calibration as automated calibration will break and fans will go full speed.  This is maybe an unrelated issue. I will look further into it. 

Appreciate your work, I hope you get some well deserved recognition 

Thank  you again for your fast help, this will help me setup my  DAWorkstation ASAP.

All the Best

closed 2025-02-17 by Snakeoilsnaker

Alcolawl, 2025-02-17:
Look for a FanControl update in the near future that will alleviate much of the issues you're seeing with sensor pairing and calibration. If you look at that same issue thread, the FanControl dev discusses with us a solution for those problems.

Thanks for all the feedback, glad it's (mostly) working for you!