Alcolawl/LibreHardwareMonitor #21: Add support for MSI PRO B840-P WIFI (MS-7E57)

seleniumsxd, 2025-02-17:
Hello, here is the LHM report for the MSI PRO B840-P WIFI motherboard.
The latest build (2522) does not appear to work with this motherboard.

[LibreHardwareMonitor.Report.txt](https://github.com/user-attachments/files/18825237/LibreHardwareMonitor.Report.txt)

Thank you for all your hard work,
I am happy to test any new builds.

Alcolawl, 2025-02-17:
Thanks for the logs! Give this build a shot and report back. Please read the README as it'll probably tackle most of the issues and oddities you might run into.

[LHM_Build_seleniumsxd.zip](https://github.com/user-attachments/files/18831210/LHM_Build_seleniumsxd.zip)

seleniumsxd, 2025-02-17:
Thanks for the quick response, everything seems to be working now!

closed 2025-02-17 by seleniumsxd