Alcolawl/LibreHardwareMonitor #22: Inclusion in LHM Afterburner Plugin

Clozic, 2025-02-17:
Hello,

thank you for all the work. As far as I have tested the system fans are not recognized by this plugin:

[LibreHardwareMonitor Afterburner Plugin](https://github.com/ts-korhonen/LibreHardwareMonitorAfterburnerPlugin)

My guess is that the plugin was build based on the standard LHM release which does not include this fix for certain MSI motherboards. However, I cannot confirm this for sure.

I have no idea how much effort is required to implement this fix into the plugin. Since it's for Afterburner (and RTSS), I expect that only the RPM sensor readouts are neccessary, not the fan speed manipulation.

Alcolawl, 2025-02-18:
Hmm. Looks like it utilizes the LibreHardwareMonitor.dll, not the LibreHardwareMonitorLib.dll that we're releasing here. Odd. Not sure if this is going to be possible from my end but I'll look into it.

Alcolawl, 2025-03-20:
Closing due to the inability to fulfill the user's request. It appears the plugin relies on different code to get working, which is not the aim of this fork.

closed 2025-03-20 by Alcolawl