Alcolawl/LibreHardwareMonitor #24: Alcolawl fix for X870 detects fans but fails in Fancontrol

3dRikal, 2025-03-15:
Somehow I can add the files to librehardwaremonitor but when I do the same in Fancontrol, it seems to work 25% of the time. 

Not sure if this is a detect problem from files or a Fancontrol issue as the files I replaced were present in Fancontrol and come from librehardwaremonitor.

HW used:
MSI X870 Gaming Plus
AMD 9900x
Fans are controlled by Corsair iCue Link & Mobo directly
XFX 7900XTX Merc310

Fans are detected by MB, by SignalRGB and LibreHardwareMonitor if I replace with Alcolawl files but Fancontrol sees only 7900XTX fans.

Alcolawl, 2025-03-20:
Other users have confirmed that if you're running some of the latest BIOS versions that it breaks Fan Control. I have a feeling this may be the case for you as well, especially if you've updated your BIOS recently. Check out the comments at the [bottom of this link](https://github.com/Alcolawl/LibreHardwareMonitor/issues/13).

chrisgleissner, 2025-03-20:
MSI has now officially released a fix for the fan control which they broke in the previous BIOS. The BIOS release (7E47v1A4) for my X870 Pro came out the other day. I have not tried it yet as I am on older BIOS, but their release notes state that they fixed fan control. 

I see that they still have to release it for your motherboard, but hopefully won't take too long. I heard of users who directly raised support cases with MSI and got a download link with a fix for their respective motherboards, so you may want to try that in the meantime.

BTW @Alcolawl , I learned the other day that the creator of Argos Monitor found a fix for the slow response times of the fans, see his release notes "Fix slow response time for fan control on MSI X870 / X870E / Z890 mainboards (Nuvoton NCT6687D-R SuperIO monitoring chip, CPU and PUMP channels)."

It may not be a fix for the slow system fans as he mentions only cpu and pump channels. I have not tried his software yet as it is not free and I am happy with my current setup. Thanks again for your work to get the system fans supported.

chrisgleissner, 2025-03-20:
Out of curiosity, I tried the latest version 7.2.02 of Argus Monitor. Whilst it is able to detect all fans (CPU fan, pump fan, 6 system fans) of my MSI X870 Pro motherboard, the speed at which it changes the RPM of my 6 system fans is about the same as when I use the patched version of LHW of this repo. The RPMs drop by about 25 RPM per second. Thus, it does not appear as though the author of that tool has found a special trick that would allow for faster reaction of the system fans.

Globespy, 2025-03-27:
> Out of curiosity, I tried the latest version 7.2.02 of Argus Monitor. Whilst it is able to detect all fans (CPU fan, pump fan, 6 system fans) of my MSI Pro X870-P WIFI motherboard, the speed at which it changes the RPM of my 6 system fans is about the same as when I use the patched version of LHW of this repo. The RPMs drop by about 25 RPM per second. Thus, it does not appear as though the author of that tool has found a special trick that would allow for faster reaction of the system fans.
> 
> Here's a screenshot for anyone interested.
> 
> <img alt="Image" width="626" src="https://private-user-images.githubusercontent.com/3969147/425201849-c56d1704-435f-40dd-8a77-bfa1d5a0b355.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3NDMwOTM4MzYsIm5iZiI6MTc0MzA5MzUzNiwicGF0aCI6Ii8zOTY5MTQ3LzQyNTIwMTg0OS1jNTZkMTcwNC00MzVmLTQwZGQtOGE3Ny1iZmExZDVhMGIzNTUucG5nP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI1MDMyNyUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyNTAzMjdUMTYzODU2WiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9MTA3Mzc3ZGQ3N2QyNzQ3ZTkzZGFjNjUxYWRiZWFmNWJiM2IxZjJlYzdjMjE3MWQyZjZkMjQ1NmVkZjVhMGZlYSZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QifQ.xhVo2OmWRrWvuQ_V-Gvo1fbYHSBNR6QrTLt7qCswCGs">

Funny! I purchased this years ago and stopped using it when FanControl came along, but of course doesn't seem to play nice with X870 boards. Wonder how the ArgusMonitor guy resolved it?

chrisgleissner, 2025-03-27:
Yeah, as it is closed source, that is hard to tell. But from what I have seen, his fan control is just as fast as the one which Alcolawl created as part of this fork. 

The main benefit I see of ArgusMonitor over FanControl is that it has a feature to log various metrics indefinitely, similar to the commercial version of Hwinfo, and that it supports the new MSI boards without any tinkering. But the first is not a deal breaker to me and the latter will hopefully resolved before long, when the PR of Alcolawl gets merged to LHW and a new release is cut.

Globespy, 2025-03-27:
I ended up just using the BIOS. I have MSI Center installed for Mystic (3rd party options have issues with bricking RAM sticks especially G.Skill), and could use their 'Cooling Wizard' but the BIOS does exactly what that does and that's one less thing to install.
Plus the BIOS is always there, software can crash.

Alcolawl, 2025-04-14:
Closed for inactivity.

closed 2025-04-14 by Alcolawl