Alcolawl/LibreHardwareMonitor #28: No speed sensors for SYS_FAN headers - MSI PRO Z890-S WIFI PZ

Mapantz, 2025-07-11:
Hi there

As per my post on the FanControl git;

CPU_FAN and PUMP_FAN speeds are being detected, but not the SYS_FAN headers.

I've changed PWM to DC for my PUMP_FAN header (there's an issue with this Lian Li pump) so there won't be a reading there anyway, I don't think?!

[image]

If there's anything else you need, please let me know.

Many thanks.

Alcolawl, 2025-07-12:
PUMP_FAN header should still report back RPMs regardless of signal method unless the pump simply doesn't report RPMs. Idk, sometimes pumps are weird.

Anyways, thanks for the screenshot, I think it gave me what I needed for a test build. Give the attached build a shot. **Read the README** for installation instructions, please. Report back on how things work!

[LHM_NCT6687DR_Build2571.zip](https://github.com/user-attachments/files/21196870/LHM_NCT6687DR_Build2571.zip)

Mapantz, 2025-07-12:
> Apologies, I did not realize LHM has changed since the last time I worked with it. I was able to replicate your issue on my end. The Zip I provided was missing some extra DLLs. Try this one.
> 
> [LHM_NCT6687DR_Build2572.zip](https://github.com/user-attachments/files/21197363/LHM_NCT6687DR_Build2572.zip)

FanControl initialized this time, but still getting this for the SYS_FANS

[image]

Alcolawl, 2025-07-12:
Manually pair the controls to the speed sensors.

Mapantz, 2025-07-12:
It doesn't do anything.

I tried pairing the sensors manually and set them all to 100%, but they don't react.

Mapantz, 2025-07-12:
[image]

Mapantz, 2025-07-13:
[LibreHardwareMonitor.Report.txt](https://github.com/user-attachments/files/21201666/LibreHardwareMonitor.Report.txt)

Thank you for looking in to this.

Alcolawl, 2025-07-14:
I'm getting reports from other users that this latest build works. Please try it with LHM first by replacing the LHM files with the files from the ZIP. Confirm that the system fan speed readouts are working. If so, then continue with replacing the files in FanControl with the files from the ZIP and test that. 

You will have to create a new configuration, calibrate the fans (may have to manually tweak them), and the fans will respond slowly (that's the case for all current gen MSI motherboards) at about 1-2% per second.

[LHM_NCT6687DR_Build2573.zip](https://github.com/user-attachments/files/21221024/LHM_NCT6687DR_Build2573.zip)

Mapantz, 2025-07-14:
> I'm getting reports from other users that this latest build works. Please try it with LHM first by replacing the LHM files with the files from the ZIP. Confirm that the system fan speed readouts are [...]

Yep!

RPM's for all of my SYS_FAN headers now showing.

Alcolawl, 2025-07-14:
Awesome! Glad to hear it. Sorry for the lengthy back and forth, there. But glad I could help.

Mapantz, 2025-07-14:
> Awesome! Glad to hear it. Sorry for the lengthy back and forth, there. But glad I could help.

No problem. It's very much appreciated!

I presume the dev of FanControl would need to update it so that automatic pairing would work?

Thanks again for your time and help.

Alcolawl, 2025-07-15:
> I presume the dev of FanControl would need to update it so that automatic pairing would work?

The dev of FanControl updates LHM that ships with the releases from time to time. Next steps for me is to create a Pull Request for the changes I made in my fork so that it's merged into LHM's codebase, then the FanControl dev can do his thing. So it may take a while. For now, stay on this version of FanControl.

closed 2025-07-16 by Alcolawl