Alcolawl/LibreHardwareMonitor #29: MSI "EZ Conn" Connected Fans not Appearing

Shnig20, 2025-07-17:
Hey,

As per conversation on [#1644 ](https://github.com/LibreHardwareMonitor/LibreHardwareMonitor/pull/1644) please see following details:

- x870 Gaming Plus WiFi Motherboard.
- The ["EZ Conn"](https://www.msi.com/blog/how-to-use-msi-exclusive-ez-conn-connector-and-cable) uses the JAF_1 header on the motherboard. I have 1 P14 max attached to it.
- It shows up in HWInfo not LHM.
- I also tried adding the sensor via the HWInfo pluggin
- I imported the latest ZIP in #1644 to both my Fan Control folder and my LHM folder both of which are running the Net 4.X version.
- Please find attached Screenshots and report as requested.

[image]
[image]

[LibreHardwareMonitor.Report.txt](https://github.com/user-attachments/files/21295115/LibreHardwareMonitor.Report.txt)

Thanks for looking into this!

Alcolawl, 2025-07-17:
Thanks for all the information so far. There are 4 things I need to identify before I can begin writing code for you to test:

- Control Register (How we talk to the fan header)
- 7-Point Fan Curve Registers (Where we store how fast the fan should be spinning)
- Fan Speed Register (How fast the fan is spinning)
- Fan Duty Cycle Register (PWM %, how hard are we making the fan work?)

I've identified the first two bullet points, the second is more challenging without having my own test system, so I must rely on you for some extra information.

Would you mind going into the BIOS and setting your EZ-Connect Fan to a manual, flat value of say 40 or 60%? Let me know which one you choose. Then, boot into Windows, check HWInfo to note the RPM of the Arctic P14 Max you have connected to that header and include the value here, along with a screenshot. Additionally, attach a new LHM report. This would help me greatly. If I know how fast the fan is spinning and what duty cycle it's operating at, it'll help me narrow down the location of the last two registers. We can't control a fan if we don't know how fast it's spinning.

Shnig20, 2025-07-17:
Ok, sorry about the delay, I disconnected the fan because it was melting my head being stuck at 50% in fan control. It's connected again. So I set the EZ Conn to a flat value of 40% and the RPM of the P14 max is averaging at 1490 RPM, as per screenshot.

[image]

I then closed HWinfo and created another LHM report which is attached.

[LibreHardwareMonitor.Report 2.txt](https://github.com/user-attachments/files/21306333/LibreHardwareMonitor.Report.2.txt)

Alcolawl, 2025-07-18:
All good! I was busy at work and then went out to a nice dinner. Had a few drinks and was ready to come back and tackle this puzzle for a bit. 

Hopefully my prior post made sense. Thank you so much for the data! I believe it's allowed me to identify all of the proper registers to set and get data to/from to make this work but I have no way to test, so this may take a bit more back and forth to get right. Give this build a shot and report back!

[LHM_NCT6687DR_Schnig.zip](https://github.com/user-attachments/files/21308190/LHM_NCT6687DR_Schnig.zip)

Shnig20, 2025-07-19:
That sounds like a lovely evening!

So, I'm very impressed! I replaced the files in LHM and fan control with the files in the zip and it's now showing up in LHM!

[image]

I presume you are well aware but the files in your zip only worked with the net 4.x of fan control 228 not the net 8.x version.

So in the Net 4.x version the EZ conn fan is showing up. however it is very buggy, it only sometimes responds to Fan Control. I tried a few different things but I couldn't tie down when and why it sometimes responds to fan control and when it doesn't. One thing I could pin down is when you load a configuration EZ fan doesn't respond.

[image]

In any event I took a few calibration points that will hopefully help you.

[image]

However, I am not convinced the numbers are correct as this is [doesn't tie in with Arctics numbers](https://www.arctic.de/media/8a/18/01/1710507321/Spec_Sheet_P14_Max_EN.pdf).

Hopefully this helps somewhat!

I also have a LHM report for you. Let me know what else you need!

[LibreHardwareMonitor.Report 3.txt](https://github.com/user-attachments/files/21329728/LibreHardwareMonitor.Report.3.txt)

Alcolawl, 2025-07-19:
@Shnig20 

Yeah, so a couple of things to address, there. 

First off, my bad. I should've made you aware that LibreHardwareMonitorLib only has a 4.72/4.8 release. Glad you got that figured out, though.

Secondly, I forgot to mention that fan control of any system fan, including the EZ-Connect fan is going to be slower than most users expect. It will completely disregard the speeds set in FanControl and only obey the fan response times you have set in the BIOS. I noticed you had the system fans set to 0.1s ramp up/down times in the BIOS for the system fans (couldn't find the EZ-connect one, it just reads as 01 in the register and that appears hundreds of times all over the place), I'd recommend setting the EZ-Connect header to the same speed for maximum responsiveness. Even so, the fastest a fan will respond is about 1-2% per second. I have yet to find a way around this and MSI refuses to provide documentation. That may play into what you're seeing here.

You're also on the right track with the calibration thing. Arctic claims the P14 Max spins at 2800 RPM (+/- 10%) at full tilt. It's clear that your fan is within spec when you have it attached to a "normal header". Unfortunately, despite the FanControl dev's best efforts, it's still difficult to calibrate fans on these motherboards automatically and I always recommend doing it manually. Manually set the fan to 100% on the EZ-Connect header and wait a minute (like actually 60 seconds) for it to reach maximum speed, then add the calibration point. Lower it to 90%, wait 30 seconds, rinse and repeat. That should give you a good calibration. Now, if the fan refuses to spin to ~2700 RPM at 100% on the EZ-Connect Header but does all day long at a normal header, that is super interesting and something I'd have to dive into further but I can't make any promises. 

Hopefully this all makes sense and gives you a good sense of direction. Let me know how it goes.

demomanca, 2025-07-20:
Just retracing my steps on all this (haven't been in the slack group for ages). Wasn't the ezconn fan controlled by the "old" setup ok? as in, same as pump and CPU? (I think it was fan "4" on the original stack)

Shnig20, 2025-07-20:
Are

> Just retracing my steps on all this (haven't been in the slack group for ages). Wasn't the ezconn fan controlled by the "old" setup ok? as in, same as pump and CPU? (I think it was fan "4" on the original stack)

If I understand you correctly, (and I'm not convinced I am, so apologies in advance) I think you're thinking of the 3-1 cable that controlled the vrm fan, radiator fans, and pump? That's a separate thing which I'm not using. The ez conn is a proprietary MSI 7 pin header on the motherboard labeled JAF1. The motherboard comes with a 2 (fan connector and RGB connector) into 1 cable that connects into that. I don't have the RGB half of that 2 into 1 cable connected.

Alcolawl, 2025-07-21:
> Just retracing my steps on all this (haven't been in the slack group for ages). Wasn't the ezconn fan controlled by the "old" setup ok? as in, same as pump and CPU? (I think it was fan "4" on the original stack)
I'm not entirely sure. I had no way to test it. I referred to the Google Sheet you and I put together during our original foray into all of this and I think most of the EZ-Conn stuff at the time was based solely on the work you had done. I've since added to it, identifying some more stuff but apparently it's not good enough yet?

> If I understand you correctly, (and I'm not convinced I am, so apologies in advance) I think you're thinking of the 3-1 cable that controlled the vrm fan, radiator fans, and pump? That's a separate [...]

I think he was more asking me about the fan control registers during our work together on this effort.

> No, don't be silly. I was fairly sure that was going to be the case, but thanks anyway!
> 
> I have set the ram up times for all fans in the bios the same now but most of the time the EZ connect just sits at 60%, completely unresponsively regardless what you have fan control or the bios settings at. Sometimes it kicks back into doing what Fan Control tells it but those bouts of response are not regular or triggered consistently. That said when it was responsive I was able to calibrate it but it's gone back to 60% again now, as is it's whim.

You say that the fan does not obey your inputs, regardless of the BIOS settings. I just have to ask this, though I'm sure you've tried already: Have you set the EZ-conn fan in the BIOS back to Smart or whatever instead of a flat, static value?

The fact that the fan is responding _sometimes_ is encouraging, meaning that the fan curve registers we identified are the correct ones, as well as the command registers, but it looks like one of two things might be happening:

1. There is some kind of conflict between FanControl and the BIOS or another piece of software (I would hope you don't have any other monitoring or fan control programs running simultaneously)
2. The code I wrote isn't consistently commanding the fan and/or re-writing the fan curve.

However, perhaps those drinks with dinner weren't the greatest move, because after reviewing some of our documentation, it looks like @demomanca had the control register for the EZ-Connect Fan documented already and I must've just breezed right over it. In the easiest way I can possibly explain, the CPU and PUMP fans are controlled traditionally, much like motherboards of the past. It's also why they spin up and down much faster than the system fans. The method I used to control the system fans is basically setting the "Smart" fan curve to a flat value, so that the fan has no choice but to spin at the value the user or FanControl specifies. This is because we couldn't find registers that controlled the System Fans like the CPU and PUMP fans are. I basically assumed that EZ-Connect would also play by those rules, but it looks like there is a corresponding control register for the EZ-Connect. Sure as shit, I reviewed your latest LHM report and the register (0xA2B) is set to a Hex value of 99, which translates to.... 60%. 

So, here's a new build, where the EZ-Connect fan obeys the logic of the CPU and PUMP fans instead of the Fan Curve brute force method. I even managed to spell your username right this time around. Let me know if this works better!

[LHM_NCT6687DR_Shnig_v2.zip](https://github.com/user-attachments/files/21351132/LHM_NCT6687DR_Shnig_v2.zip)

Shnig20, 2025-07-21:
Class, you're a quick operator. Seems to be working as expected now! Anything you want me to test/check?

Apologies @demomanca, my suspicion about misunderstanding you were correct!

Alcolawl, 2025-07-21:
Sweet!

I'd say just give it a test drive for a day of normal use. Make sure you've got your fan curve dialed in, etc. and just observe when you can and verify that the fan you have connected to the EZ-Connect header is behaving normally. Report back and if everything is good to go, I'll create a new release on my fork for the public to use and get a pull request going to have it merged into LHM itself.

Thanks for your patience, swift and legible communication during all of this. Without people like you, everyone else would be having a bad time controlling their fans on MSI motherboards this year.

Alcolawl, 2025-07-21:
@Shnig20 

Actually, mind giving this latest build a shot? I noticed a logic error in a conditional statement I wrote. This should still work but avoid any weird behavior and/or bugs.

[LHM_NCT6687DR_Shnig_v3.zip](https://github.com/user-attachments/files/21354505/LHM_NCT6687DR_Shnig_v3.zip)

Shnig20, 2025-07-22:
> [@Shnig20](https://github.com/Shnig20)
>
> Actually, mind giving this latest build a shot? I noticed a logic error in a conditional statement I wrote. This should still work but avoid any weird [...]

So I tried this and every time I wake the computer from sleep I get the following error. Tried restarting and reinstalling etc it happens every time computer sleeps.

[image]

Shnig20, 2025-07-22:
Sorry, good point/question! I just tested with v1 and v2 in this thread (uninstalled>deleted config file>reinstalled>extracted your zips into the program folder>reloaded backed up config), and yes same issue with awaking from sleep in all three versions in this thread.

Ha, good idea!

Alcolawl, 2025-07-22:
@Rem0o Could you lend some insight on what might be happening here? This sounds like a tornado of Windows Sleep and Driver nonsense, FanControl trying to restore functionality, and possible LHM having a really bad time when Windows resumes from sleep.

Alcolawl, 2025-07-22:
> FanControl basically fetch the number of sensors LHM detects. Then after sleep, it calls Computer.Close() then Computer.Open() then checks if the number of sensors is the same. If it's not, it will call Close()/Open() a few times as a retry mechanism.

So what @Shnig20 is most likely seeing here isn't a result of something FanControl is trying but rather LHM having a hard time dealing with the OS resuming from sleep, correct?

Shnig20, 2025-07-22:
I feel like an idiot here, because that sensor error wasn't happening at restarts or opening the programme I didn't think it was anything to do with my saved config. Then when I went to completely rule that out I did discover that other distracting bug that I ended up trying to trouble shoot so I never got to the bottom of my saved config question. I just reloaded that saved config and changed the sensors it was looking for (excluded PSU etc.). This seems to have refreshed the sensors in my saved config and now the waking issue isn't happening any more. Sorry for not isolating that earlier!

Shnig20, 2025-07-22:
So unfortunately the wild ride isn't over. That sensor issue is back again. Seems to re-appear after restarts. 

> I then tried to re calibrate again but the ez conn fan was now not responding to my commands or manual control, stuck at 10% this time. I tried a fresh installation again and this time it wasn't even letting me control the ez con fan to calibrated it. Just wasn't responding at all. Then after a bit more troubleshooting (me banging my head off the table while installing it and uninstalling etc) I loaded a backed up config low and behold it responded again, then i deleted configs without doing an uninstall and it was responding again even when doing new config. So basically I have no idea what that particular issue is, sometimes it responds in a new config, sometimes it doesn't.

This issue is also still there.

I'm changing some of my case fans tonight anyway. When calibrating the new ones I will just do a clean install of fan control and  import from a configuration rather than just loading it and see what happens!

demomanca, 2025-07-22:
> @demomanca Do you have an EZ-Connect Fan that you'd be able to test? I'm not sure how you intially identified the EZ-Conn registers.

I do, gimme a bit. 

If I remember correctly, it's at what we called the "fan 4" register (with 1 being cpu, 2 being pump) and it should be controlled with the "old" way.

demomanca, 2025-07-22:
Works to control EZ-Conn in LHM latest release, but breaks all motherboard sensors in fancontrol.

demomanca, 2025-07-22:
7/23/2025 8:12:10 AM: Open - System.MissingMethodException: Method not found: 'System.Security.AccessControl.FileSecurity System.IO.FileInfo.GetAccessControl()'.
   at LibreHardwareMonitor.Hardware.KernelDriver.Install(String path, String& errorMessage)
   at LibreHardwareMonitor.Hardware.Ring0.Open() in C:\Users\tdean\Documents\Git\LibreHardwareMonitor_Alcolawl\LibreHardwareMonitorLib\Hardware\Ring0.cs:line 49
   at LibreHardwareMonitor.Hardware.Computer.Open() in C:\Users\tdean\Documents\Git\LibreHardwareMonitor_Alcolawl\LibreHardwareMonitorLib\Hardware\Computer.cs:line 504
   at FanControl.Domain.BackendProviders.LHM.LHMBackendProvider.Open()
7/23/2025 8:12:15 AM: Sensor validation failed.

Alcolawl, 2025-07-22:
That is bizarre. I've tested the build on my end and everything behaves the same, but I don't have an EZ-Connect Fan. So, controlling an EZ-Connect fan breaks everything else? What could possibly cause that?

I added an extra fan and extra control to the code, added the proper registers to the fan header arrays, and modified the logic in Set6687DRControl() to only write the fan curves for the system fans and control the CPU, Pump, and EZ-Connect fan the traditional way.

Before:
```
        else if (chip is Chip.NCT6687DR) // MSI AM5/LGA1851 Motherboards
        {
            FAN_PWM_OUT_REG = new ushort[] { 0x160, 0x161, 0xE05, 0xE04, 0xE03, 0xE02, 0xE01, 0xE00 }; // Duty Cycle Sensors
            FAN_PWM_COMMAND_REG = new ushort[] { 0xA28, 0xA29, 0xC70, 0xC58, 0xC40, 0xC28, 0xC10, 0xBF8 }; // Control Registers for CPU/Pump, Initial Fan Curve Registers for System Fans
            FAN_CONTROL_MODE_REG = new ushort[] { 0xA00, 0xA00, 0xA00, 0xA00, 0xA00, 0xA00, 0xA00, 0xA00 };
            FAN_PWM_REQUEST_REG = new ushort[] { 0xA01, 0xA01, 0xA01, 0xA01, 0xA01, 0xA01, 0xA01, 0xA01 };
        }

    private void Set6687DRControl(int index, byte? value)
    {
        if (index > 1) // System Fan Control
        {
            int initFanCurveReg = FAN_PWM_COMMAND_REG[index];       // Initial Register Address for the Fan Curve
            int targetFanCurveAddr = initFanCurveReg;               // Address of the Current Fan Curve Register we're writing to
            ushort targetFanCurveReg;                               // Integer value of the current fan curve register address, not the value within
            byte currentSpeed = ReadByte(FAN_PWM_OUT_REG[index]);   // Current Speed of the target fan
        
            // If current fan duty cycle matches requested duty cycle, skip re-writing the fan curve 
            if (currentSpeed == value.Value)
            {
                return;
            }
            else
            {
                // Write 7-point fan curve
                for (int count = 0; count < 14; count += 2)
                {
                    targetFanCurveAddr = initFanCurveReg+count;
                    targetFanCurveReg = Convert.ToUInt16(targetFanCurveAddr);
                    WriteByte(targetFanCurveReg, value.Value);
                }
            }
        }
        else // Control CPU and Pump Fan normally
        {
            WriteByte(FAN_PWM_COMMAND_REG[index], value.Value);
        }
    }
```

After:
```
        else if (chip is Chip.NCT6687DR) // MSI AM5/LGA1851 Motherboards
        {
            // Each index in the below arrays represents a fan header
            // ARRAY_KEY = new ushort[] {CPU FAN, PUMP, SYSFAN1, SYSFAN2, SYSFAN3, SYSFAN4, SYSFAN5, SYSFAN6, EZ-CONNECT FAN};
            FAN_PWM_OUT_REG = new ushort[] { 0x160, 0x161, 0xE05, 0xE04, 0xE03, 0xE02, 0xE01, 0xE00, 0x163 }; // Duty Cycle Sensors
            FAN_PWM_COMMAND_REG = new ushort[] { 0xA28, 0xA29, 0xC70, 0xC58, 0xC40, 0xC28, 0xC10, 0xBF8, 0xA2B }; // Control Registers for CPU/Pump/EZ-Connect Fan, Initial Fan Curve Registers for System Fans
            FAN_CONTROL_MODE_REG = new ushort[] { 0xA00, 0xA00, 0xA00, 0xA00, 0xA00, 0xA00, 0xA00, 0xA00, 0xA00 };
            FAN_PWM_REQUEST_REG = new ushort[] { 0xA01, 0xA01, 0xA01, 0xA01, 0xA01, 0xA01, 0xA01, 0xA01, 0xA01 };
        }

    private void Set6687DRControl(int index, byte? value)
    {
        if ((index > 1) && (index < 8)) // System Fan Control
        {
            int initFanCurveReg = FAN_PWM_COMMAND_REG[index];       // Initial Register Address for the Fan Curve
            int targetFanCurveAddr = initFanCurveReg;               // Address of the Current Fan Curve Register we're writing to
            ushort targetFanCurveReg;                               // Integer value of the current fan curve register address, not the value within
            byte currentSpeed = ReadByte(FAN_PWM_OUT_REG[index]);   // Current Speed of the target fan
        
            // If current fan duty cycle matches requested duty cycle, skip re-writing the fan curve 
            if (currentSpeed == value.Value)
            {
                return;
            }
            else
            {
                // Write 7-point fan curve
                for (int count = 0; count < 14; count += 2)
                {
                    targetFanCurveAddr = initFanCurveReg+count;
                    targetFanCurveReg = Convert.ToUInt16(targetFanCurveAddr);
                    WriteByte(targetFanCurveReg, value.Value);
                }
            }
        }
        else // Control CPU, Pump, or EZ-Connect Fan normally
        {
            WriteByte(FAN_PWM_COMMAND_REG[index], value.Value);
        }
    }
```

Alcolawl, 2025-07-22:
This may have something to do with it
https://github.com/Alcolawl/LibreHardwareMonitor/commit/06b714ae1b1a09fdbe866a5bd718bbdbb8fa5f55

My fork is 9 commits behind including the one above. Let me re-base my fork and put my changes back in, build again and I'll host it here.

demomanca, 2025-07-22:
Also, I'd be careful because LHM uses the order of the PWM Out reg list against the control mode and pwm request reg, so because it's fan "4" it'll need to be in the 4th spot, or, we need to force it to tell it it's fan 4

Alcolawl, 2025-07-22:
[LHM_NCT6687DR_Build2574.zip](https://github.com/user-attachments/files/21376425/LHM_NCT6687DR_Build2574.zip)
Try that build and see if it works better. Rem0o's commit to LHM seems to be pretty critical and it was missing in the most recent builds posted in here.

> Also, I'd be careful because LHM uses the order of the PWM Out reg list against the control mode and pwm request reg, so because it's fan "4" it'll need to be in the 4th spot, or, we need to force it to tell it it's fan 4

I've assigned EZ-Connect as the last spot in every instance can be assigned. It's always the 9th fan (index 8). I've set it as such in SuperIOHardware.cs and nct677X.cs. That's why the Set6687DRControl function will treat fan a fan with an index > 1 and < 8 as a system fan, where index 0 is CPU, 1 is Pump, and 8 is EZ-Conn... Unless I'm not really grasping what you're saying, in which case, apologies.

demomanca, 2025-07-22:
All g man. So the EZ-Conn fan will need to be Index 4, because the manual control flag at A00 uses the index to tell which bit to flip to control the fan. Gotta duck out for a few hours, but I'll try that when I get back.

Alcolawl, 2025-07-22:
> All g man. So the EZ-Conn fan will need to be Index 4, because the manual control flag at A00 uses the index to tell which bit to flip to control the fan. Gotta duck out for a few hours, but I'll try that when I get back.

Well then we might have a problem there since there are 9 fans and only 8 (0 through 7) bits that can be flipped.

[image]

If the function gets fed an Index value of 8, that bitwise math is gonna come out super funky, which would explain why things start misbehaving when controlling the fan. Because there are only 8 bits in a byte... The only thing I can conclude is that there is either a separate register with a bit that needs flipping for proper EZ-Connect control...

demomanca, 2025-07-23:
Also, did your board not come with the adapter to use that port?
It's got a fan, argb and a usb 2 header split on it. 

[image]

Alcolawl, 2025-07-23:
> Na, the sysfan control logic you build doesn't use those bits in A00, only the "old" fans 1-8 that always worked before we got involved. Remember, this was the issue we hit, we couldn't find the 0xA00 equivalent for the "back 8" fans. I haven't looked at what's there now, but basically I listed all 16 controls. RPMs etc, and then just selected fan 1,2,4,11,12,13,14,15,16 etc. I'd just be using your control logic for the back 6 fans.

Ah, it's literally in the comments for the NCT6687D above. lol
[image]

I think I see what you're saying now, though. The old code for the 6687D has 16 Fans and Controls, but before I tackled this EZ-Conn request, we were only using 8 Fans, so I cut the array down to 8. This way, when someone opened up LHM or FanControl, they didn't have 14 System Fan entries of which only 8 were functional. You're saying I'd have to reinstate the old 16 index array and ensure that the EZ-Connect Fan Registers were at Index 4. This way, when we go to control it like the CPU and Pump fan (using the traditional method) it can flip the bit properly and we skip that if it's a system fan?

Also, I'm sure my motherboard came with that connector, just never looked into EZ-Conn too much because I assumed it required a special compatible fan... lmao.

Shnig20, 2025-07-23:
Nope, normal pwm fan attached to their cable which attaches to that ez conn header.

Is there anything I can do for you at my end?

Alcolawl, 2025-07-24:
> Nope, normal pwm fan attached to their cable which attaches to that ez conn header.
> 
> Is there anything I can do for you at my end?

Not currently, no. :( @demomanca and I are trying to work it out offline, so we're not spamming this issue thread but **we are** working on it. The challenge right now is the legacy code was written to handle 8 fans. It did so by flipping a bit in a certain 8-bit register. 8-bits. Now we have 9 fans. CPU, Pump, 6 System, and EZ-Connect. There's no 9th bit, obviously. So more research or trickery has to be done. However, we've since discovered that this bit flipping stuff doesn't need to be done when using my brute force method for controlling the system fans.

For shits and giggles, give this build a shot and let me know if it works. I'm trying to handle the EZ-Connect fan like a system fan again, but I reset my fork to include Rem0o's commit that I referenced the other day and that should help with some of the errors you were seeing, at least.

[LHM_NCT6687DR_Shnigv4.zip](https://github.com/user-attachments/files/21416343/LHM_NCT6687DR_Shnigv4.zip)

Shnig20, 2025-07-25:
Hey,
Thanks for those. I didn't have much time today so I only very briefly tested V4 but appeared to be very similarly inconsistent to V3 couldn't pin down bugs etc but the same bugs appeared to be present.

The latest one you have compiled I spent a bit more time with. It appears to be working well! I'm guessing because of the way you changed how to handle fans the sensors were all over the place when I initially loaded it but when I did a clean install and imported from the configuration as opposed to loaded and paired the speed sensors and calibrated them everything appears to be working as intended. I let the computer sleep, woke it and restarted system ect. Everything looks fine.

Here's a screenshot and report.

[LibreHardwareMonitor.Report 5.txt](https://github.com/user-attachments/files/21431909/LibreHardwareMonitor.Report.5.txt)
[image]

The system fans still don't automatically pair to speed sensors but I think that is an issue with all versions? Ez Conn fan does automatically pair to speed sensors btw.

Apart from that I also noticed that while in bios you can put all the sys fans into different groups except the EZ Conn fan. I don't know does that give you any clues?

Also slightly off topic but how would you recommend to set the fans in bios for best practice for use with Fan Control?

Thanks for sorting this. I'm very impressed with the work!

closed 2025-07-25 by Alcolawl

Alcolawl, 2025-08-12:
@Shnig20 

I haven't really pursued it much, work and life things have gotten in the way. Additionally, merging my EZ-Connect changes into LHM Master would break every FanControl user's configurations if they're using a motherboard that takes advantage of the NCT6687DR workaround. This would create a nightmare for the LHM maintainers and FanControl dev. The only viable way I see around this is creating my own fork and builds of FanControl as well, which, selfishly, I don't want to pursue, I'm sitting in front of a computer enough as it is and the amount of issues that would flood my way would be nothing but a headache. 

The only other viable alternative to the things mentioned above is maintaining my own fork, as I do now, and let technically savvy users take advantage of it. This way, I'm not breaking tons of people's functional configurations and creating more headaches for everyone involved. This is unfortunately the result of having designed a workaround while mainly ignoring the EZ-Connect fan header. My fault.

As for not upgrading FanControl, you can still do so, no need to stay on an old version, you just have to replace the files in the FanControl folder with the files in the zip of the build I have you (or build 2572) each time. That's what I do. You'll get all of the new features of FanControl, if there are any in the update, while maintaining the modified LHM functionality.