Alcolawl/LibreHardwareMonitor #33: Missing RPM support for MSI PRO B850M-P WiFi (control works, tach not exposed)

jackoregankenny, 2025-09-06:
Hi maintainers,

Thank you for all the work you put into LibreHardwareMonitor — it’s been fantastic to see new boards and sensor chips being supported.

I think there may be a small gap in the recent MSI B850 support. In commit 0A7534A (v25.2.2), the MSI PRO B850-P WiFi was added. My board is the MSI PRO B850M-P WiFi (µATX), which uses the same Nuvoton NCT6687D/R SIO controller.

Here’s what I observe with FanControl v234 and LibreHardwareMonitor 25.2.2:

Fan RPM sensors are not exposed — FanControl reports “no speed sensor” for SYS_FAN headers.

In BIOS and HWiNFO, RPM values for SYS_FAN headers are visible and update correctly.

This makes me suspect that the B850M-P WiFi variant string might not have been included alongside the B850-P WiFi, even though the hardware is effectively the same.

Details:

- Board: MSI PRO B850M-P WiFi (µATX)
- Chipset: B850
- SIO: Nuvoton NCT6687D
- Fan headers tested: CPU_FAN, Pump_FAN, SYS_FAN1/2
- BIOS: Latest
- OS: Windows 11 Pro (latest)
- FanControl: v234
- LHM: 25.2.2 (commit 0A7534A / 8138C03)

Expected: SYS_FAN RPM sensors exposed the same way as on the MSI PRO B850-P WiFi.
Actual: SYS_FAN control works, but tach sensors are not mapped.

If adding the B850M-P WiFi identifier is all that’s needed, I’m happy to test any build that includes it. Thank you again for maintaining this project — it’s incredibly useful and I really appreciate the ongoing support for newer hardware.

jackoregankenny, 2025-09-09:
Absolutely, I was fluting around trying to add it myself (and failing miserably). Here's what I do have, I'm no longer using your recent version so the control isn't appearing but this should have it? 

[image]

Any other way I can help lmk. I've tried lots. HWinfo and the bios have no issue, haven't tested Argus monitor yet.

Alcolawl, 2025-09-09:
Perfect, thank you. Adding support for a motherboard is actually pretty easy if you're curious about learning. 

Here's a commit where I added support for 3 additional motherboards: https://github.com/Alcolawl/LibreHardwareMonitor/commit/3b1c1cbd123184e2dfc6d38bee0410b4c3ca2fb3

As you can see, all I had to do was amend 4 different files. I'll list and explain each one below:

1. Model.cs - Basically a big list of Motherboard models. You need to add a new MSI entry here for the specific motherboard. This will serve as a shorthand reference for the motherboard for everything downstream.
2. Identification.cs - Identifies the motherboard by looking at the name it grabs from the register **and associates it with the new model entry you created in model.cs** (This is why I needed the screenshot. I needed "MSI PRO B850M-P WIFI (MS-7E71)").
3. LpcIO.cs - Sets the chip object to "NCT6687DR" instead of the standard "NCT6687D" SIO controller when the model string you created in Model.cs matches the ones listed in the switch case.  This effects downstream functions for Fan Control and identification where the "D-R" variant is handled differently.
4. SuperIOHardware.cs - Sometimes optional. Can be used to dictate custom sensor layouts in LHM vs the standard one. This usually only applies to the higher end boards that have more gizmos on them.

Anyways, that's basically what I've done for your board in the build below. Please remember to look at the README and follow the instructions and report back when you've got it running. Let me know how it works and if there are any issues. Also, if you have any further questions on the code changes I've made to get all this working, I'll be more than happy to share. 

[LHM_NCT6687DR_jackoregankenny.zip](https://github.com/user-attachments/files/22242028/LHM_NCT6687DR_jackoregankenny.zip)

jackoregankenny, 2025-09-09:
Hell yeah thanks for this! It seems to work amazingly - thank you so much 

Gonna try a few boards we have in work and add them too! Thanks for teaching me!

closed 2025-09-09 by jackoregankenny

Alcolawl, 2025-10-26:
Hold on, let me get the timeline straight, because the way I'm currently interpreting it makes me think this isn't an LHM related issue...

1. So you were running FanControl v236 with presumably with the LHM build I built for you on September 9th.
2. Suddenly it stops working.
3. You upgrade to a newer version of FanControl, which relies on a newer build of LHM that requires the PawnIO driver and sensors and controls are still broken?

Did you install the [PawnIO driver](https://pawnio.eu/)? Did you replace the FanControl LHM files with the files from the latest release of this fork? I'm a little lost on what went wrong and when, but I've been running the latest release with FanControl v244 (and earlier versions since the PawnIO switch) and it's been flawless.

jackoregankenny, 2025-10-26:
Sorry yeah I could have been clearer. That's my bad @Alcolawl  

Running the build you had generated September 9th. This worked until this week. It suddenly stopped despite no updates or changes. 

I first tried to fix it. Re-installing the fix you very kindly provided with the existing version. Both as a direct swap for LHM or via plugin. (Not sure why it felt like the right way)

Following this I saw your new release and tried it with the new version of fan control. Also dropped it direct into LHM. 

Both of these could not see the tachometer. 

I've since tried a few combos of different versions. Your build for me specifically the new build. Making sure to swap in and out the versions of fan control as appropriate. 

I've installed, uninstalled. Deleted the folder for both fan control and tried PawnIO install directly too. I also varied the "smart fan" mode in the bios. Trying DC, PWM and smart.

The one thing I didn't try actually was a fresh install of the newest with PawnIO. Dropping your new fork in and restarting my pc before use.

Alcolawl, 2025-10-26:
Actually, here's an updated build with 20+ upstream commits and my changes I hate waiting in my local repo, including the fixes for your specific motherboard above. I've tested it and it's running perfectly on my end with FanControl v245.

[LHM_NCT6687DR_Build25102RC.zip](https://github.com/user-attachments/files/23153042/LHM_NCT6687DR_Build25102RC.zip)